prusa3d/PrusaSlicer #4786: M600 transition point should be covered by color change

Charlieray77, 2020-09-22:
### Version
2.2.0

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
Makerfarm i3v10" custom marlin 2.0.5.3

### Behavior
M600 color transition point occurs outside of the area that the next color will cover, so there is a color blob of the next color after the filament change and before the hot end moves to the area that will be printed next.
I wish the hot end would do this color change in an area that will be covered by the next color filament. Otherwise I have to scrape off a blob.

This would be a new feature request.

#### Project File (.3MF) where problem occurs
[USPS Heart v1.zip](https://github.com/prusa3d/PrusaSlicer/files/5263727/USPS.Heart.v1.zip)

rtyr, 2020-09-23:
This is not a task for PrusaSlicer. M600 command is handled by printer firmware. Closing.

closed 2020-09-23 by rtyr

Charlieray77, 2020-09-23:
The SLICER determines WHERE the M600 happens.  IT should happen AFTER the head moves to the new location of the new color print, change color and then print.

This is a problem in PRUSASLICER.

rtyr, 2020-09-23:
>I wish the hot end would do this color change in an area that will be covered by the next color filament. Otherwise I have to scrape off a blob.

Actual color change is made on the spot specified by the firmware, M600 just triggers it.

You probably meant this https://github.com/prusa3d/PrusaSlicer/issues/4432.