prusa3d/PrusaSlicer #4787: Feature Request: Add custom Gcode section to print settings for simpler implementation of M907

FilamentFrenzy, 2020-09-22:
[image]
Request for a “Custom Gcode” section within the “Print Settings” tab, just the same as exists currently under “filament settings” and “printer settings” tabs to make implementation of the M907 simpler and more free and understandable  to stock and none stock Prusa profiles.

rtyr, 2020-09-23:
I don't think such modification is needed. M907 command is used only with MK3 profiles and it may be removed with some future firmware update. It is quite uncommon to use such command in the start g-code.

rtyr, 2020-12-16:
I believe we can close this.

closed 2020-12-16 by rtyr