prusa3d/PrusaSlicer #4788: Warn about overlapping objects when slicing

PetteriAimonen, 2020-09-23:
### Version
_PrusaSlicer version 2.2.0+linux+x64_

### Operating system type + version
_Ubuntu 18.04_

### 3D printer brand / version + firmware version (if known)
_Original Prusa i3 MK3S_

### Behavior
This is partly an user error, but I think it would be nice to have a warning about this as it is an easy mistake to make.

1. Open PrusaSlicer and switch to `Preview [Ctrl+6]` before adding any objects. (Alternatively: switch to preview after slicing an object and select `Edit -> Delete all` to remove it.)
2. Load a new object using menu: `File -> Import -> Import STL`. Note that the preview does not get updated, I guess this is a small bug also.
3. Go selecting settings. Because the preview didn't get updated, you forget that you already added the object.
4. Add the object again using `File -> Import -> Import STL`. Press `Slice now` and everything proceeds as normal.
5. Printer ends up extruding twice the amount of plastic as is desired, with terrible results.

I would want that when I press `Slice now`, PrusaSlicer would pop up a warning `You have overlapping objects, are you sure you want to proceed?`.

Like shown below, the problem is not visible in preview or 3d editor unless one drags the object with mouse.

[image]
[image]
[image]

foreachthing, 2020-09-23:
And if I do that on purpose, then I get a warning about something I want? Could it be a exclamation mark in the statusbar? I really, really don't like popups!

PetteriAimonen, 2020-09-23:
Yeah, I guess an exclamation mark near the slicing status output or even a line like "Overlapping objects: 2" could work.

neophyl, 2020-09-23:
I opened this issue https://github.com/prusa3d/PrusaSlicer/issues/3678 about having an opacity/xray mode on the platter.  With such a mode you would be able to tell from the opacity if there were multiples over each other as well as the primary purpose of placing modifiers inside objects more easily.  2 birds with one stone.  Without a pop up :)

rtyr, 2020-09-23:
Also related to this issue https://github.com/prusa3d/PrusaSlicer/issues/4520.

edwardhorsford, 2022-01-11:
I was just bitten in a print by not noticing an overlapping part - a soft warning in the lower right would be great.

In my case I'd used `+` to duplicate a part many times. Unfortunately this part was small and ended up overlapping a larger part so I couldn't see it easily. Only when printing did I notice two parts had ended up merged.

If arrange let you arrange only items on the build plate I would use that and avoid this issue - but it arranges everything, including stuff off the build plate I don't want to currently print. The current solution of picking items manually isn't ideal if you can't see they're overlapping, nor if you have many items.

DangeMask, 2022-11-29:
Today I had similar issue. I wanted to print a multipart model as one piece (no need to screw together for me), so I just overlapped the objects in slicer, roughly checked the preview and hit print. Then I realised the printer prints the overlapping parts twice. So I had to clancel the print, merge the objects in slicer and start again.
It just wasn't obvious to me that the slicer would behave this way. I believe a notification or exclamation mark would help with this. Maybe use the same way how slicer suggests automatic color changes for signs. It can also contain a quick link to "merge objects" which would help people who just want to merge objects and don't know about double printing of overlaps.

PS: I wanted to use "Add part" at first, which would solve this too. But i needed to cut some parts at first and the slicer only allows cut for the whole boject, not one part of it.

ats1995, 2023-07-07:
For me an option to highlight the parts that overlap and warn would be super helpful when fitting parts densely on the build plate.

closed 2025-03-06 by Jan-Soustruznik