prusa3d/PrusaSlicer #4791: [BUG] - Generated support with 0.01mm height ?

gedeonlab, 2020-09-23:
### Version
PrusaSlicer v2.2.0 win64

### Operating system type + version
Windows 10 19041.508

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S stock model. Version of the firmware v3.9.0-3421

### Behavior
* _Describe the problem_
the  **generated support** (enforced in this case) as several **layers of 0.01 height** so it's hard (impossible) to print and generate clicking when the printer try to extrude support. The general layers height is at 0.07.

* _Steps needed to reproduce the problem_
I use a stl model, cut it to keep a small part and add some enforced support. I also see same issue with other models (see 3mf and stl file in zip attached)

* _Expected Results_
for me support layer height should be not smaller than the perimeter layer height or not less than 0.05mm.

* _Actual Results_
for exemple between layer 120 and 121 the layer height is 0.01 (first picture) and other layers are same height (second picture)
[image]
[image]

#### Project File (.3MF) where problem occurs
[bugslicingsupport0p01.zip](https://github.com/prusa3d/PrusaSlicer/files/5270210/bugslicingsupport0p01.zip)

rtyr, 2020-09-23:
I checked your 3MF. There are no support layers printed with 0.01mm layer height. It is just side effect of the preview, where are combined support layers and object layers, each printed with different layer height.

You can easily check it if you enable only support layers in the preview (see screenshot).
[image]
[image]

gedeonlab, 2020-09-24:
ok, you're right in the support vue layers has 0.07 layer height.

So I don't understand why extruder only clicking on support layer and not on perimeters layer when it print this bunch of layer. maybe clicking on support of 0.07 layer height....

thanks for your answer

closed 2020-09-24 by gedeonlab