prusa3d/PrusaSlicer #4792: STL file moves after pressing "Slice Now"

Reconfirefly, 2020-09-23:
### Version
PrusaSlicer-2.2.0+win64-202003211145

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 5 Plus

### Behavior
I have created a 350 x 350 mm rectangular bed to use prusaslicer with my E5+, but when I click slice now to slice the part, it ends up being shifted significantly in the direction of the red axis arrow (X or Y, not sure). This bug does not appear for the default prusa, or with the ender 3 preset. See the attached gif.
[image]

lukasmatena, 2020-09-24:
Please, post a 3MF.

Reconfirefly, 2020-09-26:
[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/5286279/3DBenchy.zip)

There you are. I did a little more investigating, and it turns out that these lines in the starting gcode are the culprits...
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line

lukasmatena, 2020-09-26:
Thanks. The issue only affects the preview, not the gcode. And the culprit are not the intro lines, but the line just after it:
`G92 E.0.0 ;`. The extra dot after the E confuses the gcode preview so it shows the whole print shifted.

lukasmatena, 2020-09-26:
> The issue only affects the preview, not the gcode

Just to clarify that, I mean that the gcode extrusions have correct coordinates. The syntax error in G92 is probably interpreted by the gcode preview to set current nozzle position as zero (hence the shift depending on intro lines). I don't know what the printer firmware will do with it. Maybe the same, maybe not.

Reconfirefly, 2020-09-28:
That was the cause. Removed the extra period and all is working as it should.

closed 2020-09-28 by Reconfirefly