prusa3d/PrusaSlicer #4793: "Not responding" every time opens on 2nd monitor

jagr200, 2020-09-24:
PrusaSlicer 2.2.0
Windows 10 Pro x64

So this seems relatively new but when I open PS on my second monitor it always goes into a "not responding" state. It only loads the frame and the top bar but never shows the bed layout. It worked last week but this week it has been failing. Not sure what info I can pull to make this easier to figure out because I'm not very good with that stuff. Anyone else experience this?

rtyr, 2020-09-24:
Looks like:
#2522
#2142
#3689

Nahimic driver issue?

jagr200, 2020-09-25:
> Looks like:
> #2522
> #2142
> #3689
> 
> Nahimic driver issue?

Thank you so much. I didn't see those because I didnt know what nahimic was hahaha. That fixed it. Why would I just start having that issue? Maybe a game I just installed caused it?

kocikdav, 2021-02-01:
Hello. I would like to ask users having troubles with Nahimic libraries to try this build. It should display warning dialog about NahimicOSD.dll linked.
https://www.dropbox.com/s/on0dwea75gfx4cc/PrusaSlicer-2.3.0%2B127-win64-gf939cac84-202102011345.zip?dl=0

We know that this is just a warning rather than a solution, but currently we have no way to stop loading this library as it comes linked statically when we start our process.

Also if you are aware of another libraries of same source please let us know their names. You can find it with ProcessExplorer tool.

bubnikv, 2021-07-30:
Warning about the Nahimic DLL being injected into PrusaSlicer will be rolled out with PrusaSlicer 2.4.0-alpha1.
It is not a "fix" for the nahimic issue, but at least the user will be informed.

closed 2021-07-30 by bubnikv