prusa3d/PrusaSlicer #4797: Bridging wrong direction

dheera, 2020-09-24:
### Version
2.2.0
### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
PrusaSlicer tries to bridge the wrong direction in the first bridging layer. It should bridge the short direction in the first bridge layer.
[image]
[image]

ochm, 2020-10-03:
Try to change brifging angle value.

dheera, 2020-10-08:
@ochm That fixed it, thanks!
I guess then, this issue gets converted to a feature suggestion: auto-detect the optimal bridging angle for each bridge, since there may be multiple bridges that have different optimal angles in the same print.

Pele79, 2020-11-18:
When you roll your cursor over this setting it says "If left to zero, the bridging angle will be calculated automatically" so yes it should automatically choose the best option but it doesn't seem to change imho. I have attached 2 images of the same model and bridge or overhang actually, but it really highlights the issue, both images were taken with the bed in the same orientation, I have just rotated the model 90 degrees.

As you can see although the main bridging angle is not completely parallel to the longer edge, the angle is still very shallow:
[image]

With the model rotated it is much batter, but the angle in relation to the bed is either the same or very close:
[image]

Hopefully this is something that can be looked at.

foreachthing, 2020-11-18:
Could you please attach your file?
I just noticed, that with this build (#feffb66), my part bridges much nicer:
[image]

Pele79, 2020-11-18:
Like I say what I have shown is actually an overhang to a support, but its unsupported.  Due to what I can upload I have had to zip it, it will load standing on the narrow end but I just dropped it on its side.

[Hoek_LACK_hoog.zip](https://github.com/prusa3d/PrusaSlicer/files/5562374/Hoek_LACK_hoog.zip)

foreachthing, 2020-11-18:
Yeah, when I rotate your part like so:

[image]

It actually looks good... what do you think?

Pele79, 2020-11-19:
I like that the lines are not at an angle, but surely it would better that the bridging lines should run along the Y-axis not the X-axis, as the Y is the shortest distance to bridge, if you see what I mean.

foreachthing, 2020-11-19:
Well, then you'd need to add a modifier - a bridging-angle modifier.
[Hoek_LACK_hoog_2.zip](https://github.com/prusa3d/PrusaSlicer/files/5566781/Hoek_LACK_hoog_2.zip)

1. Right mouse click on model
1. add modifier - slab, box, .... I've used a slab
1. click the gear of the generic slab
1. add settings - infill
1. choose bridging angle; set value to 90° or adjust for your needs

danieltwagner, 2024-03-11:
I'm new to PrusaSlicer and just encountered this. I never had this happen with Cura and having to inspect my bridges before printing because I can't trust the defaults isn't a great experience, even though I otherwise very much enjoy PrusaSlicer.
I see this hasn't been touched in a few years. Is this because it's hard to do or because it hasn't been deemed sufficiently important? Would you accept a patch for this, assuming I can find my way through your code base?

closed 2026-09-29 by prusaslicer-bot[bot]