prusa3d/PrusaSlicer #4798: Missing rectangular infill edges when printing -  they appear in slicr

kevinmichaeldudley, 2020-09-25:
### Version
_Version of PrusaSlicer used goes here_ 2.2.0

_Use `About->About PrusaSlicer` for release versions_2.2.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
Rectangular infill lines NOT printing but shown in slicr
* _Steps needed to reproduce the problem_Print the file
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ Yes
Appears fine in slicr but edges are missing when printed.  Refer to images.
And , no, the printer is not goofing up, unless it "magically" does it exactly in the same location on every layer.
Do blobs no strings just NO PRINTING of those edges.  looks to me like it's relative to the outer perimeter of the inner hole object.

[image]
[image]

#### Project File (.3MF) where problem occurs

[K0064SB 44T Stud Gear V5_.zip](https://github.com/prusa3d/PrusaSlicer/files/5284539/K0064SB.44T.Stud.Gear.V5_.zip)

rtyr, 2020-09-26:
The problematic lines of rectilinear infill are printed right after series of tiny gap fill extrusions/retractions (on the teeths). Such scenario can lead to decreased pressure in the nozzle. The missing/weak lines are most likely caused by that.

[image]

bubnikv, 2020-11-20:
Likely a hardware issue. Closing.

closed 2020-11-20 by bubnikv