prusa3d/PrusaSlicer #4799: Simultaneously over- and under-extruding on topmost layer causing holes and bulging at the same time

zeus86, 2020-09-25:
### Version
PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage

### Operating system type + version
Ubuntu 18.04 x86_64

### 3D printer brand / version + firmware version (if known)
custom, mostly a kossel (deltabot)

### Behavior
* first layer prints perfectly fine (no, really!), as well as the second (mostly), but the 3rd consecutive solid layer (regardless if part of the bottom or top in the part) shows various defects, which I cannot fully explain. As this affects primarily the topmost layer (because visible), i will refer to this. 
The topmost layer shows signs of underextrusion **AS WELL** as overextrusion. rectangular infill starting from a corner is very thin for the first few lines, but then slowly turns to normal extrusion. after an area of good extrusion, material bulges up in front of the nozzle, like moving a wave in front of you. Once too much excessive material has build up, the nozzle cannot keep up, producing some very bulged up layers on the surface, which exceeds upwards by more than one layer height when cooled down(!!). Thank god for spring-loaded beds... After this bulged area the infill-line continues with a thin section like at the start of the infill and the whole thing repeats. Note that this happens all in one go, in a consecutive line of infill on the top layer, so I would exclude things like retraction, travelspeed or z-hop here, because they do not apply in the middle of an infill. I have seen this defect before, but mostly on the very first layer on uneven beds when the nozzle is too close and ripping the layer apart, but not for the solid top layer. 

This one is not easy to describe, so I will add some reference pictures. the test-piece is a ~60x80x2mm plate with some holes and some tabs on the top surface, nothing special (created from OpenSCAD, so i am pretty sure it has no defects). one picture is from another version of the part with some more holes, but basically the same part. Don't mind the markings, I was playing around with parameters.

On smaller Objects, like the 20x20mm calibration cube this problem is not as pronounced like on a big, flat surface, but still there. On complex objects like a benchy the final print suffers from some layers, which gets squished out to the side leaving a "layer-wide, consecutive blob" on the outer perimeter. 

### Steps needed to reproduce the problem
unsure how to reproduce

### What was tested
I tested at reasonable speeds <50mm/s and fiddled around with extrusion multiplier (0.9 to 1.05), extrusion width (0.4 to 0.5, @ .4 nozzle) and temperature (190-210 °C, PLA)
lowering the extrusion multiplier did not result in any better parts, but the surface "sags" more into the part, making the underlying infill more pronounced (stars, 20%) (best to described as "like a stepped pillow").

### Expected Results
even top surface

### Actual Results
jagged and teared top surface, showing signs of over- and under-extrusion as well as repeating patterns of scars along with the infill-direction.
Some of the attatched pictures do not look like that bad qualitywise, but believe me, they are. the bulging is on some spots build up more than 0.3mm, so more than one layer height. 

### Note
Maybe it's worth mentioning, that im not new to 3d-printing at all, neither to deltaprinters. I know, some slic3r-settings are not finetuned yet and some others are fairly conservative (e.g. speeds), because I recently upgraded to new hardware (skr 1.4 pro with TMC 2208 and BTT TFT28). After some years not printing I teared down my old delta and build it up from scratch. Basically it's a kossel with 20x20 extrusions and a heated bed (mk2 with bonded aluminium and buildtak). filament train consists of an e3dv6 and a bondtech-bmg-clone.

### Project File (.3MF) where problem occurs
[surface-defects.zip](https://github.com/prusa3d/PrusaSlicer/files/5285024/surface-defects.zip)

### Images for reference
[image]

[image]
circles: underextrusion
squares: bulging

some more pics (yeah i know, potatocam...):
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I really know someone knows how to fix this and especially what causes this specific issue.

rtyr, 2020-09-26:
I tried it on my custom Kossel (also SKR 1.4, TMC2208, V6) and the result is fine. Only some overall overextrusion, which is expected with your custom filament diameter value. I think it is safe to assume, that the cause of your issues is not in the gcode.

Does it happen with different material than PLA? Your problem can be related to some sort of clogging/heatcreep issues. Partial clog on bowden printer can create both underextrusion and overextrusion at the same time (as the filament is "pressurized" inside the PTFE tube during the clog phase and then rapidly released). You can try the same model with PETG or ABS, which are not prone to heatcreep issues.

[image]

zeus86, 2020-09-27:
Oh awesome, you printed my part :-). if you have no further use for it: i designed this as covering for the bottom section of the kossel (if using 20x20 beams with 20mm spacing, i am currently designing some variants with holes for various plugs and lcd-voltagemeters and so on...)
If the Problem is not the gcode, are you willing to share your marlin config then? i guess this would bring up the error fastest. I am currently using the 2.0-bugfix-release, but I am not nailed to ultra-fresh features :-).

Anyhow, thank you alot for testing until here.

zeus86, 2020-09-27:
> Does it happen with different material than PLA? Your problem can be related to some sort of clogging/heatcreep issues. Partial clog on bowden printer can create both underextrusion and [...]
...forgot to mention the rest of it: 
actually i have no other material other than PLA on hand, but i do have other PLAs lying around. i just wanted to use up my old spool while calibrating. I don't assume heatcreep will be an issue, because i currently use a 40mm-noctua-fan (always on, not pwm-regulated with those adapter included), and all the mounting stuff for the fans are printed from PLA also (with my old RAMPS before upgrading, but on the same material). If heatcreep would be an issue, these parts would have been suffered from massive deformation meanwhile, but they do not. 
I clearly have have some issues with my part cooling (but i currently cannot reliably print upgraded parts for this, because of this isssue here). So best bet would be the firmware, so if you would be willing to share your's (assuming you are using marlin), this would be awesome.

Thanks in advance

rtyr, 2020-09-27:
Just some correction, I have SKR v1.3 (not 1.4). Currently I have TMC2208 only on XYZ and A4988 on the extruder. I had TMC2208 on the extruder as well for some time, without issues. I switched it to A4988 while I did some experiments with the linear advance.

Anyway, here it is (2.0.3-bugfix):
[marlinconfig.zip](https://github.com/prusa3d/PrusaSlicer/files/5287876/marlinconfig.zip)

zeus86, 2020-09-28:
I am finally back, and I also got some findings. First of all: thank you @rtyr for sharing your config. It helped me alot to make sure i did not have any drastic errors in my config. One noticeable difference was, that i don't run the TMC2208 in standalone mode, but in UART mode, because they are a) quieter, because of stealthcop, can b) be monitored by software an c) have their current beeing set in software and d) interpolation to 256x-microstepping (not really sure, if this setting apllies in standalone mode) really advisable!

### Here's what I've done:
* Although I have had LIN_ADV disabled, I still had some K-Factor beeing set in the firmware. I cannot confirm if it applies to a print, I set it to 0.
* Originally I had the TMC2208-Microstepping set to 4 (because i've read about, that this should have more limits left when running from 12V like i do). With interpolation to 256 steps this should not be a difference. I switched to 16x, because why not. My drivers are not running at thier limits whatsoever and are actively cooled. Boy what a difference. Even though they are out of the box much quieter than a4988 or drv8825, at 16x-stealthchop2-mode they are barely noticeable. another massive improvement in noise!
* I wanted to play around with junction deviation, and had set it, but on compilation it complained, that for deltas and scara kinematics classic jerk have to be used, so i reverted to classic jerk, but lowered it a bit like you did. 
* although i don't use fw-retract, because it gave me inconsistent results also, i now completely disabled it, to make sure it does not interfere with anything. 

### First test print: 
Looked very promising at first, the described effect was reduced by a fair amount, but far away to be gone, so i looked into the differences apart from the firmware side ad did further tweaks...

### Further tuning:
* too slow speeds can cause issues of its own, and even if i'm sure this wone be a problem here, i ramped up to a baseline of 60mm/s for (inner) perimeters and infill
* i disabled s_curve_acelleration, which I wanted to test, just to be sure.
* reduced (de-)retraction-speed, more retract in general

### Second test print: 
Better than the first, but still the same issue

### Further tuning:
After deeper looking into, what could cause "inconsistent extrusion", because this seems to be the underlying problem, i came across one massively overlooked adjustment, and I cannot believe that this was the Problem in the end: 

**Overtightened Extruder** ...yes, exactly that...
In my self defense: I use a BMG-Dual-Extruder-Clone, and you cannot really determine how tightened it actually is, because the screw has more or less the same resistance while screwing over almost the whole thread, and it is enclosed also. I loosened it to to point at which it barely grabs, tightened 1/2-3/4 turn and guess what: 
**The Problem is almost gone!** (some smaller signs of this effect are still there, but I am pretty sure, that printing hotter, and therefore reducing the nozzle pressure will fix this also)

### Lessons learned: 
Even if you build up 5+ printers from ground up (and not just kits), which are all working and printing fine with different kinematics, you are not immune to basic overlooks, which are willing to shoot you in the knee..

...so,,,yes, this issue can be closed than, because this issue was not slicer-related...or software-related...just me-related.

closed 2020-09-28 by zeus86