prusa3d/PrusaSlicer #4801: Feature request: Toggle STL fix and error checking off

ozzyman, 2020-09-26:
### Version
2.2.0 Linux X64

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Any (curently Snapmaker A350)

### Behavior
* _Describe the problem_

When attempting to export Gcode, an error is reported   (empty layers detected). Gcode is not exported to disc.

* _Steps needed to reproduce the problem_

Attempt to slice a model with a DELIBERATE horizontal air gap (0.08mm, 0.16, 0.24 and 0.32mm in the case of my test object) 

* _If this is a command-line slicing issue, include the options used_

* _Expected Results_

 Gcode is exported to disc.

* _Actual Results_

Gcode is NOT exported to disc

* _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

Yes. In expert mode, error checking should be able to be selectively disabled. At present, print-in-place models are difficult or impossible to slice using Prusaslicer.

#### Project File (.3MF) where problem occurs

[0.08_vertical_clearance_test.zip](https://github.com/prusa3d/PrusaSlicer/files/5286251/0.08_vertical_clearance_test.zip)

This model slices and prints using other slicers, but I want to be able to use variable layer height to speed up printing while maintaining the gap dimensions

_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-09-26:
The 'empty layers detected' error is already possible to dismiss in current master (and it will be possible in 2.3.0). See https://github.com/prusa3d/PrusaSlicer/issues/4756#issuecomment-692709951. Does that solve the issue?

ozzyman, 2020-09-28:
> The 'empty layers detected' error is already possible to dismiss in current master (and it will be possible in 2.3.0). See [#4756 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/4756#issuecomment-692709951). Does that solve the issue?

No. 

When slicing, the error dialog pops up, I click "OK", and the model slices. So far, so good. When exporting the gcode, however, gcode is not saved after clicking "OK".

lukasmatena, 2020-09-28:
Are you absolutely sure that you are running current master? That is, you have compiled the source code yourself and the title bar now reads PrusaSlicer-2.3.0-alpha0+UNKNOWN, without any reference to 2.2.0?

closed 2024-02-18 by lukasmatena