prusa3d/PrusaSlicer #4802: [feature request] Automatic skirt adjustment when using a brim

Moorviper, 2020-09-26:
### Version
2.2.0+

### Operating system type + version
doesn't matter

### 3D printer brand / version + firmware version (if known)
doesn't matter

### Behavior
the skirt is inside the brim area

[image]

[Container_50x100_NoCore_v2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5286531/Container_50x100_NoCore_v2.3mf.zip)

It would be great if the skirt adjustment would be automatic.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

TD5023, 2020-09-27:
Not that this fixes the problem, but do you even need a single-layer skirt at all? It's been awhile since I printed anything with a brim, but whenever I did, I disabled skirts entirely. If I needed extra filament priming, I just made the brim a little wider.

GaryBoone, 2021-01-10:
I can't tell if this bug is open or closed: It says Open at the top of the page, but there's a Nov 1, 2020 marking from rtyr showing it Closed?

This is clearly a bug: The brim feature does not work as intended, drawing around the print, if the brim is too wide. 

+1 to TD5023's point: In addition to fixing the bug, the UI could point out to the user that if you're using a brim, you most likely don't need a skirt. However, that assumes that the outermost brim paths are printed first, so the skirt function of ensuring even flow can occur then. If an inner path is printed that is actually part of the component being printed, then any flow issues could affect of the component. Is there a guarantee that that the brim goes outside in, or completes before any part of the component is started?

mlcdev, 2021-01-19:
I believe PrusaSlicer should at least warn about this possible issue. Solution would be to auto adjust skirt range to bring + 1mm or so...

neophyl, 2021-01-19:
The closed tag is for issue 5056 which is the one mentioned next to that tag.  This issue has had so many iterations. It’s the way it is due to user request, there are specific use case for how it is. Read through some of the older open and closed issues if you want the full history. I believe it’s been left the way it is as it is under user control. If you don’t like the skirt cutting the brim then adjust its position. As it is at the moment it doesn’t really cause a problem.