prusa3d/PrusaSlicer #4803: Feature Request: Slic3r-style 3D Honeycomb Infill

unicornswag, 2020-09-27:
Hey, so I made a litte writeup on this issue, along with some photo examples [here.](https://forum.prusaprinters.org/forum/prusaslicer/prusaslicers-3d-honeycomb-infill-is-sub-optimal/)

TL;DR: PrusaSlic3r's implementation of "3D Honeycomb" infill is less efficient than Slic3r's, mainly due to it's habit of alternating the print direction after each layer. This produces infill that consists almost exclusively of bridges, and results in relatively weak parts. In Slic3r's version or 3D Honeycomb, the print direction only alternates when it starts a new layer of "cells." This produces parts that are much stiffer in all directions, and have much better inter-layer adhesion due to the greatly-reduced number of bridges.

gringer, 2022-05-18:
This is still an issue in PrusaSlicer v2.4.2

pmarks-net, 2024-04-18:
This was recently fixed in OrcaSlicer 2.0: https://github.com/SoftFever/OrcaSlicer/pull/4425

gringer, 2026-09-03:
This is still an issue in PrusaSlicer; there have been *numerous* updates to the 3D Honeycomb code, the most obvious of which is that the pattern switches at the smallest point rather than every layer. This makes the pattern much stronger:

https://github.com/OrcaSlicer/OrcaSlicer/pull/4425#issuecomment-2002028587

This issue has been fixed in OrcaSlicer. See the `Fill3DHoneycomb.cpp` source for more details:

https://github.com/OrcaSlicer/OrcaSlicer/blob/main/src/libslic3r/Fill/Fill3DHoneycomb.cpp