prusa3d/PrusaSlicer #4805: Vase mode 'skipping' layers: rounding error?

ProfEngr, 2020-09-28:
### Version
2.2.0+win64

### Operating system type + version
Win7

### Behavior
Depending on the layer height in the Print tab, a varying number of 'missed' layers appear in the preview. These layers are not empty, but there are only a handful of gcode lines compared to thousands for normal layers. I've also seen extrusion paths crossing open air outside the printed part shell.
[image]
[image]
At 0.19LH it only missed 2 layers. That was the fewest number from 0.10 to 0.39 that I tried.

#### Project File (.3MF) where problem occurs
[pyramid6-gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/5293944/pyramid6-gcode.zip)

If the 3MF export doesn't include the Octoprint API code then I can reslice and upload it as well.

ProfEngr, 2020-09-28:
It's possibly related to the STL itself from the writeup in this link a Twitter user sent to me.
https://www.thingiverse.com/thing:2613568

rtyr, 2020-09-30:
This was already discussed here https://github.com/prusa3d/PrusaSlicer/issues/520 some time ago.

It looks fine, when i set "Slice gap closing radius" to 0.049 (which is default value at the moment).

[image]

bubnikv, 2020-12-09:
explained by @rtyr 
Closing

closed 2020-12-09 by bubnikv