prusa3d/PrusaSlicer #4806: [2.3.0-alpha0] Specific start g-code causes the software to crash

th33xitus, 2020-09-28:
### Version
PrusaSlicer-2.3.0-alpha0
Commit `7a10e23` to `f890cd5`

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Firmware: Klipper (but shouldn't matter)

### Behavior
Today i updated my local PrusaSlicer repo because i wanted to build a new alpha 0 release from source.
I encountered a really weird issue when using my start g-code i always used before (and which works totally fine in PrusaSlicer 2.2.0 and earlier alpha 0 versions).

<details>
  <summary>My start g-code looks as follows:</summary>

  ```
  ;### SET HB/HE AND WAIT
M190 S{first_layer_bed_temperature[0]}
M109 S{first_layer_temperature[0]}
;### HOME
G92 X0 Y0 Z0 E0
G28
;### LOAD MESH
BED_MESH_PROFILE LOAD=default
;### MOVE TO PURGE START
G1 E5 Z2 F720
G1 X15 Y3 F6000 ; move nozzle to purge start
G1 Z{first_layer_height} F720 ; lower nozzle to print height
;### PRIME NOZZLE
G1 X203 Y3 E{(nozzle_diameter[0]*first_layer_height*188*4)/((filament_diameter[0] * filament_diameter[0])*3.14159)} F600
;### LIFT NOZZLE -> GET READY FOR PRINT
G1 Z2 F720
G1 Y5 F6000
G92 E0.0
  ```
  
</details>

When i use that start g-code above and slice something, the most recent alpha 0 will crash without an error message.
I get the following console output when i execute it with `--sw-renderer --loglevel=9`:
<details>
  <summary> Console Output</summary>  

```
Loading MESA OpenGL library: S:\Programmier-Projekte\Slicer\PrusaSlicer-master\build\src\Release\mesa\opengl32.dll
MESA OpenGL library was not loaded
[2020-09-28 19:59:21.770758] [0x00003e6c] [debug]   full path: 8488602939018436129
[2020-09-28 19:59:21.770758] [0x00003e6c] [debug]   single instance: false. other params: S:\\Programmier-Projekte\\Slicer\\PrusaSlicer-master\\build\\src\\Release\\prusa-slicer-console.exe : --sw-renderer : --loglevel=9 : --datadir : S:\\MEGACloud\\PrusaSlicerExp\\
[2020-09-28 19:59:21.770758] [0x00003e6c] [debug]   init wx instance checker 8488602939018436129.lock S:\MEGACloud\PrusaSlicerExp\/cache/
[2020-09-28 19:59:21.771758] [0x00003e6c] [info]    instance check: Another instance not found or single-instance not set.
[2020-09-28 19:59:21.841735] [0x00003e6c] [info]    Checking if indices need to be installed from resources...
[2020-09-28 19:59:21.843735] [0x00003e6c] [trace]   translation_language provided by PrusaSlicer.ini: en_GB
[2020-09-28 19:59:21.843735] [0x00003e6c] [trace]   System language detected (user locales and such): de_DE
[2020-09-28 19:59:21.845734] [0x00003e6c] [trace]   Best translation language detected (may be different from user locales): de_DE
[2020-09-28 19:59:21.845734] [0x00003e6c] [trace]   Switching wxLocales to en_GB
[2020-09-28 19:59:22.499525] [0x00003e6c] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB WorkingSet: 51MB; PrivateBytes: 39MB; Pagefile(peak): 39MB(41MB)
[2020-09-28 19:59:25.802468] [0x00003e6c] [info]    Checking for cached configuration updates...
[2020-09-28 19:59:25.803468] [0x00003e6c] [info]    Confing bundle not installed for vendor BIBO, skipping:
[2020-09-28 19:59:25.804468] [0x00003e6c] [info]    Confing bundle not installed for vendor Creality, skipping:
[2020-09-28 19:59:25.806467] [0x00003e6c] [info]    Confing bundle not installed for vendor LulzBot, skipping:
[2020-09-28 19:59:25.808466] [0x00003e6c] [info]    Confing bundle not installed for vendor PrusaResearch, skipping:
[2020-09-28 19:59:25.809466] [0x00003e6c] [info]    No configuration updates available.
[2020-09-28 19:59:25.841455] [0x00002d00] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
*   Trying 185.115.1.124...
* TCP_NODELAY set
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* schannel: SSL/TLS connection with files.prusa3d.com port 443 (step 1/3)
* schannel: checking server certificate revocation
[... 520 lines cut ...]
;### MOVE TO PURGE START
G1 E5 Z2 F720
```
I know that sounds like it doesn't make any sense at all (i didn't want to believe it myself) but i can reproduce the crash 100% of the time when i use that exact command.

The funny thing is, that the workaround to "fix" the crash is, to write `G1 E5 F720` and `G1 Z2 F720` each in its own line.
So writing this, will actually "fix" the issue and doesn't let the software crash:
```
;### MOVE TO PURGE START
G1 E5 F720
G1 Z2 F720
```
This is also 100% reproducible in my case.

---

Now that i found out what causes the issue, i wanted to find out WHEN this issue started to appear.
I had a built available from 2020-08-21 which worked fine with the original start g-code, so it must be a commit between August 21st and today which breaks something. I made like a dozen builts to pin down the commit where the issue appears for the first time and came to the conclusion that it is https://github.com/prusa3d/PrusaSlicer/commit/7a10e23470384ba13062a059b48425b86a58b7ed which causes a crash with my original start g-code. And using `G1 E5 F720` and `G1 Z2 F720` each in its own line works fine again.

When i build with the previous commit (https://github.com/prusa3d/PrusaSlicer/commit/743d6643ae0bfa33dfdd961eb9458ac377694dd9) it does NOT crash when using `G1 E5 Z2 F720`. So that works totally fine like it always did.

So there must be a change in https://github.com/prusa3d/PrusaSlicer/commit/7a10e23470384ba13062a059b48425b86a58b7ed which causes the software crash without an error message. Whats weird is, that `G1 E5 Z2 F720` is a totally fine g-code command. I don't see why it should create any issue for the program to process it. But seems like there is one.

I hope i provided enough information for you guys to try and maybe reproduce the issue and/or fix it. 
If you need more, please tell me and i try to provide it to you.

lukasmatena, 2020-09-29:
> I hope i provided enough information for you guys to try and maybe reproduce the issue and/or fix it.

You did indeed, thanks. The call stack at the time of the crash:
```
1   std::vector<unsigned int>::size
2   Slic3r::GUI::GCodeViewer::load_toolpaths
3   Slic3r::GUI::GCodeViewer::load
4   Slic3r::GUI::GLCanvas3D::load_gcode_preview
5   Slic3r::GUI::Preview::load_print_as_fff
....
```

The crash happens at GCodeViewer.cpp:1322:
```
unsigned char id = buffer_id(curr.type);
TBuffer& buffer = m_buffers[id];
MultiIndexBuffer& buffer_indices = indices[id];
if (buffer_indices.empty())
buffer_indices.push_back(IndexBuffer());

 static const size_t THRESHOLD = 1024 * 1024 * 128;
// if adding the indices for the current segment exceeds the threshold size of the current index buffer
// create another index buffer, and move the current path indices into it
if (buffer_indices.back().size() >= THRESHOLD - static_cast<size_t>(buffer.indices_per_segment())) {
```
`curr.type` is equal to` EMoveType::NoOp`, which results in `id` getting assigned value 255, which is well out-of-range of both `m_buffers` and `indices`. The fix is not obvious to me and I will leave it to someone else (@enricoturri1966). Hopefully it will be easy to fix with the info you've given us.

n8bot, 2020-09-29:
Well, a possible fix might be this:

```diff
------------------------ src/slic3r/GUI/GCodeViewer.cpp ------------------------
@@ -31,7 +31,10 @@ namespace Slic3r {
 namespace GUI {
 
 static unsigned char buffer_id(EMoveType type) {
-    return static_cast<unsigned char>(type) - static_cast<unsigned char>(EMoveType::Retract);
+    if (static_cast<unsigned char>(type) == 0)
+        return 0;
+    else
+        return static_cast<unsigned char>(type) - static_cast<unsigned char>(EMoveType::Retract);
 }
 
 static EMoveType buffer_type(unsigned char id) {
```

lukasmatena, 2020-09-29:
@n8bot That will probably bypass the crash, but I don't know the code well enough to tell whether the NoOp is right in the first place. It may also be a manifestation of a logic error that happened earlier, such as not processing the combined G1 command correctly. This should be judged by the author or someone who does not see this for the first time.

n8bot, 2020-09-29:
Agreed. I have no idea the purpose for subtracting retract (1) from the value to begin with. Nor do I know very much about the code, or c++ in general. So, like I said, it's just a possible fix for anyone who needs it now or to help troubleshoot exactly why the value is underflowing to 255. (Is underflowing a word?)

n8bot, 2020-09-29:
@lukasmatena forgive me for continuing to discuss things I have no place discussing. If you haven't noticed, I'm using the PrusaSlicer project as a way to learn c++, and software dev in general. Thanks for all the work you guys are doing, in the open -- it's an amazing opportunity to learn from skilled devs, in an application that interests me.

I looked into the process_G1 method, (which I had coincidentally already been familiar with because it taught me what a lambda expression was) and discovered the pertinent code:

```cpp
    auto move_type = [this](const AxisCoords& delta_pos) {
        EMoveType type = EMoveType::Noop;

        if (delta_pos[E] < 0.0f) {
            type = (delta_pos[X] != 0.0f || delta_pos[Y] != 0.0f || delta_pos[Z] != 0.0f) ? EMoveType::Travel : EMoveType::Retract;
        }
        else if (delta_pos[E] > 0.0f) {
            if (delta_pos[X] == 0.0f && delta_pos[Y] == 0.0f && delta_pos[Z] == 0.0f)
                type = EMoveType::Unretract;
            else if (delta_pos[X] != 0.0f || delta_pos[Y] != 0.0f)
                type = EMoveType::Extrude;
        } 
        else if (delta_pos[X] != 0.0f || delta_pos[Y] != 0.0f || delta_pos[Z] != 0.0f)
            type = EMoveType::Travel;

        return type;
    };
```

It would appear to me that Noop is the default move_type. The previous fix I applied handled the default case, where it was un-handled before (clearly.) It's plainly obvious how the move is not being captured by any of the move types. Had the E amount been negative, i.e., less than zero, it presumably would have been captured as a retract. This particular move, which includes Z movement and a positive E amount, should I guess be classified as an "unretract" move This could be achieved like this, I presume:

```diff
-------------------- src/libslic3r/GCode/GCodeProcessor.cpp --------------------
@@ -1381,7 +1381,7 @@ void GCodeProcessor::process_G1(const GCodeReader::GCodeLine& line)
             type = (delta_pos[X] != 0.0f || delta_pos[Y] != 0.0f || delta_pos[Z] != 0.0f) ? EMoveType::Travel : EMoveType::Retract;
         }
         else if (delta_pos[E] > 0.0f) {
-            if (delta_pos[X] == 0.0f && delta_pos[Y] == 0.0f && delta_pos[Z] == 0.0f)
+            if (delta_pos[X] == 0.0f && delta_pos[Y] == 0.0f)
                 type = EMoveType::Unretract;
             else if (delta_pos[X] != 0.0f || delta_pos[Y] != 0.0f)
                 type = EMoveType::Extrude;
```

It doesn't seem like it would be problematic to ignore the Z 'word' in the case of no X/Y movement but positive E movement. It's clearly an outlier, but it is what it is.

lukasmatena, 2020-09-30:
@n8bot Thanks for looking into it. You are probably right that this is the place where the combined move goes unhandled, although I don't know the reasons and ramifications. The gcode viewer may not be even capable of rendering an extrusion with non-const z, so maybe the classification as NoOp is ok and the crashing on generating the respective toolpath is not. Again, @enricoturri1966 should have the final word. But your investigation will probably make the job a lot easier.

enricoturri1966, 2020-09-30:
@n8bot
This issue should be fixed by 980fe77b32e8730ce08173af793af53653d1dd05.
Could you please confirm ?

th33xitus, 2020-09-30:
@enricoturri1966 I just compiled a release with https://github.com/prusa3d/PrusaSlicer/commit/980fe77b32e8730ce08173af793af53653d1dd05 and i can tell, that it fixed the issue. Thank you!

lukasmatena, 2020-09-30:
@th33xitus and @n8bot 
Thanks to you both.
Closing.

closed 2020-09-30 by lukasmatena

n8bot, 2020-09-30:
Cool! I was considering the idea that the move would/should be also/instead classified as travel, with a ternary operator similar to the above "retract" case-capture. I guess visualizing an unretract with any amount of x/y/z motion could be problematic. Thanks for letting me play along!