prusa3d/PrusaSlicer #4808: [2.3.0-alpha0-13dcc3e] Negative origin-bed offsets cause "toolpath outside print area" warning

FHeilmann, 2020-09-29:
### Version
_Version of PrusaSlicer used goes here_

`version_2.3.0-alpha0-935-g13dcc3e7a`

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

Windows 10 build 19041

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Not relevant

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Not relevant

### Behavior
* _Describe the problem_
I use a negative X and Y offset for my bed shape, to make the gcode line up with the center of my bed properly. The first move to the part after the gcode therefore starts outside the printable area and causes a "toolpath outside print area was detected" warning to be thrown while slicing.
* _Steps needed to reproduce the problem_
- Set bed with negative X and Y offsets
[image]
(start gcode for reference)
[image]
- Slice any object
* _Expected Results_

Slicing process succeeds without warnings

* _Actual Results_
[image]
(my best guess as to why the warning was generated circled in green

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[clotheshanger_fix.zip](https://github.com/prusa3d/PrusaSlicer/files/5298868/clotheshanger_fix.zip)

enricoturri1966, 2020-09-30:
@FHeilmann
This issue should be fixed by: e55a9cceecda11eefa9c898324470e0230c73d7f
Please confirm

FHeilmann, 2020-09-30:
@enricoturri1966 can confirm issue is fixed on latest master (including e55a9cc) 

Thanks for the quick fix!

-Florian

closed 2020-09-30 by FHeilmann