prusa3d/PrusaSlicer #4810: Misleading title for "Bridge flow ratio"

haleypearse, 2020-09-30:
I always thought this was a flow ratio, but it's in fact **Bridge extrusion width**. Flow ratio preserves the x,y,z axes, which this does not.
[image]

Set to 0.1: 
[image]

Set to 2 (max): 
[image]

Set to 0.005 :) 
[image]

rtyr, 2020-09-30:
Bridge extrusions are round and their diameter is equal to the nozzle diameter. You can't control the width if you extrude in the air.

See https://manual.slic3r.org/advanced/flow-math for more details.

haleypearse, 2020-09-30:
Interesting. The "flow ratio" affects the position offset of the bottom surface of the part. This might help with my original problem, not being able to decrease the unusable 0.2mm support distance. Also reminds me of a feature I've long wanted: Bridge z-offset. But you're right that this is more than an extrusion width. It's extrusion diameter. Flow ratio unfortunately doesn't show in the viewer or affect speeds limited by max volumetric speed.

flow ratio: 1, 0.1mm layer height:
[image]

flow ratio: 0.2, 0.1mm layer height:
[image]

flow ratio: 0.2, 0.1mm layer height:
[image]

rtyr, 2020-09-30:
As for the support, the bottom most object layer is printed with a bridging flow, it is intended and it was always like that. You can use negative values for z contact distance.

rtyr, 2020-10-01:
Can we close this issue?