prusa3d/PrusaSlicer #4811: "External perimeters first" modifier adds internal perimeters

Ofenhed, 2020-09-30:
### Version
2.2.0

### Operating system type + version
Windows 10

### Behavior
* I want to add subtle recycling symbols on my prints (a manual version of #4349). I attempted at doing this by adding a modifyer to my print with the option "External permimeters first", to create seams around my symbol.
* The expected result is that the outer layers that the modifier comes in contact with would be inverted.
* The actual result is that internal perimeters are created.
[image]

#### Project File (.3MF) where problem occurs
[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/5304317/Shape-Cylinder.zip)

(The recycle symbol was created by [Makerten](https://www.thingiverse.com/Makerten) at [thingiverse](https://www.thingiverse.com/thing:3191698) under Creative Commons - Attribution license)

Ofenhed, 2020-09-30:
This seems to be the case for many perimeter settings, such as changing perimeters count as well.

rtyr, 2020-09-30:
The way the modifier volumes work is they split the object into pieces, which are sliced independently, so it is expected behavior (at least at the moment).

Duplicate of #3846, #3317, #3742, #4201. Closing.

closed 2020-09-30 by rtyr