prusa3d/PrusaSlicer #4813: Long pause "Generating Index Buffers" during slicing

pmjdebruijn, 2020-09-30:
### Version
2.3.0~alpha0+953~g8bf0f75e8 (current master)

### Operating system type + version
Ubuntu Linux 20.04

### 3D printer brand / version + firmware version (if known)
Modified Ender 3

### Behavior
When slicing a fairly small and simple model with current master, I get a very long pause (minutes) with the message "Generating Index Buffers" on a fast system:

[image]

enricoturri1966, 2020-10-01:
@pmjdebruijn
Could you please check if 8ded9dc0fdb80305955cd52a81b0e4c2eaaa9af8 helps in improving the progress dialog performance ?

pmjdebruijn, 2020-10-01:
Yep, seems much better.

Though I'm not sure a dialog is the right choice considering UI design. For one, it completely blocks the UI, not sure if that's entirely intentional...

Maybe a progress bar directly above the 'Slice now' / 'Export G-Code' button would be better?

pmjdebruijn, 2020-10-25:
Both the original issue and my suggestion for improvement have been resolved. Thanks!

closed 2020-10-25 by pmjdebruijn