prusa3d/PrusaSlicer #4814: Retraction is set to 80mm and it won't let me change it

Nathan116, 2020-10-01:
/### Version
2.2.0

### Operating system type + version
Windows X64

### 3D printer brand / version + firmware version (if known)
TronXY X5s running Marlin 2.0.1

### Behavior
* _Describe the problem_
Retraction seems to be stuck at 80mm and it's ridiculous. I never set my retraction at 80mm (my retraction speed is 80mm/s, but my retraction length was 8.5mm). It is REALLY annoying. I have uninstalled and reinstalled PS, changed printer profiles, swapped STL files I was slicing, still the same. This started happening only after I install the latest "upgrade" today (9/30/2020).

* _Steps needed to reproduce the problem_
Literally just slice any model with any setting, it still made my retraction length at 80mm

* _If this is a command-line slicing issue, include the options used_
I don't think so, I only use the GUI side

* _Expected Results_
Well, I expected that, when I changed the retraction settings it would change the sliced value in the gcode, but it didn't

* _Actual Results_
Retraction distance was still set to 80mm

* _Screenshots from __*PrusaSlicer*__ preview are preferred_
Screenshot of the GCODE from PrusaSlicer
[image]

Screenshot of slicer settings and .ini file
[image]

neophyl, 2020-10-01:
Just as an experiment what happens if you set your retraction speed to say 20mm/s ?  If it changes to that value then at least you will have narrowed down the issue, the parser using the wrong setting.

th33xitus, 2020-10-01:
@Nathan116 Make sure you don't have any filament overwrites enabled for retraction.

I wasn't able to reproduce that issue on 2.2.0

rtyr, 2020-10-01:
We are unable to help you without any data, please provide your 3MF project file. My guess is that you have 80mm retraction set in filament overrides.

Nathan116, 2020-10-01:
> @Nathan116 Make sure you don't have any filament overwrites enabled for retraction.
> 
> I wasn't able to reproduce that issue on 2.2.0

> We are unable to help you without any data, please provide your 3MF project file. My guess is that you have 80mm retraction set in filament overrides.

You both nailed it. Had 80mm override in my filament settings (a setting I didn't even realize was there). All is fixed now

closed 2020-10-01 by Nathan116