prusa3d/PrusaSlicer #4816: Build Failure

pmjdebruijn, 2020-10-01:
### Version
`2.3.0~alpha0+971~g0bdfb6ab9` (current git master)

### Operating system type + version
Xubuntu 20.04 

### Behavior
```
[ 80%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoFdmSupports.cpp.o
cd "slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/obj-x86_64-linux-gnu/src/slic3r" && /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_LOCALE_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBNEST2D_GEOMETRIES_clipper -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK__ -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/obj-x86_64-linux-gnu/src/platform" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/clipper" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/polypartition" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/hidapi/include" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/obj-x86_64-linux-gnu/src/libslic3r" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/libnest2d/include" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/include" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/miniz" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/glu-libtess/include" -I"slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/qhull/src" -isystem /usr/include/eigen3 -isystem /usr/lib/x86_64-linux-gnu/wx/include/gtk3-unicode-3.0 -isystem /usr/include/wx-3.0 -isystem "slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/libigl" -isystem /usr/include/OpenEXR  -g -O2 -fdebug-prefix-map=slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fext-numeric-literals -Wall -Wno-reorder -pthread -O3 -DNDEBUG -fPIC   -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -pthread -DOPENVDB_ABI_VERSION_NUMBER=6 -std=gnu++17 -Winvalid-pch -include "slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/obj-x86_64-linux-gnu/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx" -o CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoFdmSupports.cpp.o -c "slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/GUI/Gizmos/GLGizmoFdmSupports.cpp"
slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/GUI/Gizmos/GLGizmoRotate.cpp: In constructor ‘Slic3r::GUI::GLGizmoRotate3D::RotoptimzeWindow::RotoptimzeWindow(Slic3r::GUI::ImGuiWrapper*, Slic3r::GUI::GLGizmoRotate3D::RotoptimzeWindow::State&, const Slic3r::GUI::GLGizmoRotate3D::RotoptimzeWindow::Alignment&)’:
slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/GUI/Gizmos/GLGizmoRotate.cpp:223:11: warning: unused variable ‘button_width’ [-Wunused-variable]
  223 |     float button_width = imgui->calc_text_size(_(button_txt)).x;
      |           ^~~~~~~~~~~~
In file included from slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/libslic3r/PrintBase.hpp:14,
                 from slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/libslic3r/SLAPrint.hpp:5,
                 from slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/GUI/Gizmos/GLGizmoSlaSupports.cpp:20:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/GUI/Gizmos/GLGizmoFdmSupports.cpp: In member function ‘virtual void Slic3r::GUI::GLGizmoFdmSupports::on_render_input_window(float, float, float)’:
slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/GUI/Gizmos/GLGizmoFdmSupports.cpp:99:114: error: cannot convert ‘__gnu_cxx::__alloc_traits<std::allocator<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char> >::value_type’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
   99 |         const float cursor_type_combo_width = std::max(m_imgui->calc_text_size(wxString::FromUTF8(cursor_types[0])).x,
      |                                                                                                                  ^
In file included from /usr/include/wx-3.0/wx/memory.h:15,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/pchheader.hpp:95,
                 from slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/obj-x86_64-linux-gnu/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx:5,
                 from <command-line>:
/usr/include/wx-3.0/wx/string.h:1742:42: note:   initializing argument 1 of ‘static wxString wxString::FromUTF8(const char*, size_t)’
 1742 |     static wxString FromUTF8(const char *utf8, size_t len = npos)
      |                              ~~~~~~~~~~~~^~~~
slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/GUI/Gizmos/GLGizmoFdmSupports.cpp:100:114: error: cannot convert ‘__gnu_cxx::__alloc_traits<std::allocator<std::__cxx11::basic_string<char> >, std::__cxx11::basic_string<char> >::value_type’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
  100 |                                                        m_imgui->calc_text_size(wxString::FromUTF8(cursor_types[1])).x)
      |                                                                                                                  ^
In file included from /usr/include/wx-3.0/wx/memory.h:15,
                 from /usr/include/wx-3.0/wx/object.h:19,
                 from /usr/include/wx-3.0/wx/event.h:16,
                 from /usr/include/wx-3.0/wx/app.h:19,
                 from slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/src/slic3r/pchheader.hpp:95,
                 from slic3r-prusa-2.3.0~alpha0+971~g0bdfb6ab9/obj-x86_64-linux-gnu/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx:5,
                 from <command-line>:
/usr/include/wx-3.0/wx/string.h:1742:42: note:   initializing argument 1 of ‘static wxString wxString::FromUTF8(const char*, size_t)’
 1742 |     static wxString FromUTF8(const char *utf8, size_t len = npos)
      |                              ~~~~~~~~~~~~^~~~

```

mkyral, 2020-10-01:
I can confirm the same issue on Gentoo.

lukasmatena, 2020-10-01:
Thanks for the report. This is caused by using wxWidgets 3.0, which lack the `wxString::FromUTF8(const std::string&)` overload.

Commit f1c24e6 should fix it. Closing.

closed 2020-10-01 by lukasmatena