prusa3d/PrusaSlicer #4817: Infill not appearing in preview mode

r0xANDt0l, 2020-10-01:
### Version

Version 2.2.0 + win64

### Operating system type + version

Windows 10

### 3D printer brand / version + firmware version (if known)

Creality Ender 3 with mods on the extruder, bltouch. and 32-bit mainboard with TMC 2209. And Marlin 2.0.6

### Behavior
* _Describe the problem_

Infill not appearing when checking the sliced model

* _Steps needed to reproduce the problem_

Slice an stl with any kind of infill

* _Expected Results_

A model sliced with infill

* _Actual Results_

[image]

#### Project File (.3MF) where problem occurs

[Pen holder.zip](https://github.com/prusa3d/PrusaSlicer/files/5314876/Pen.holder.zip)

rtyr, 2020-10-01:
Your model is hollow. Use solid model, if you want some infill inside.

rtyr, 2020-10-03:
Behavior explained. Closing.

closed 2020-10-03 by rtyr