prusa3d/PrusaSlicer #4818: Invalid wipe tower for 0.3mm layer height MMU prints when supports are enabled

iamgnat, 2020-10-02:
### Version

2.2.0

### Operating system type + version

Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)

MK3S + MMU2S

Firmware is not relevant for this issue as this is pre-gcode generation.

### Behavior

When slicing a set of models for a MMU2S multi-tool print the wipe tower works as expected for an 0.2mm layer height print. If I switch to a 0.3mm layer height without supports configured the wipe tower also generates as expected. If I add supports (everywhere, from build plate only, and support enforcers only) the wipe tower that is generated is suddenly too large for the plate. I have replicated this issue regardless of the number of objects and with multiple different objects.

The expected behavior is that the wipe tower should still be an appropriate size so that it is printable.

#### Project File (.3MF) where problem occurs

##### 0.2mm layers with supports
[image]

##### 0.3mm layers without support
[image]

##### 0.3mm layers with support
[image]

##### 0.3mm with support (zoomed out for full tower view)
[image]

###### AMF project

[wipetower_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/5315691/wipetower_issue.zip)

lukasmatena, 2020-10-02:
Duplicate of #4361 (and several other issues reachable from there). It should be already fixed in `master` for 2.3.0 release.

In the meantime, use the workaround described by rtyr in #4361. I'm closing this issue.

closed 2020-10-02 by lukasmatena

iamgnat, 2020-10-02:
Thanks @lukasmatena. I did try searching, but I didn't have success. Sorry for the dupe and thanks for already having a fix in place.