prusa3d/PrusaSlicer #4819: Export Configuration Bundle doesn't work correctly

Jnesselr, 2020-10-02:
### Version
Version: 2.2.0+
Build: PrusaSlicer-2.2.0+-202003211132

### Operating system type + version
MacOS Catalina 10.15.6

### 3D printer brand / version + firmware version (if known)
n/a but it's an MK3S stock running 3.9.1

### Behavior
- PrusaSlicer fails to export configuration bundle correctly
- Go to File > Export > Export Config Bundle...
- I expect to get a file with more than 9 lines
- I get:
[image]

#### Project File (.3MF) where problem occurs
There's nothing on the plate when I do this export. I did check with a benchy on the plate and the same issue occurs.

neophyl, 2020-10-02:
Do you have any customised saved profiles that you have modified/created yourself on your system ?

Exporting a config bundle exports only the unique non system profiles I believe.  If you don't have any then it will have nothing to export.

Jnesselr, 2020-10-02:
I do not. I'm surprised it doesn't export the system profiles but a single export will. That feels counterintuitive. I'm mostly trying to make it easy for users to do an export of all the profiles and then do an import for BotQueue.

I think I'll just pull the ini files from your github. Hopefully the path to them doesn't change often.

Since this seems like a "works as intended", I'll close it.

closed 2020-10-02 by Jnesselr

neophyl, 2020-10-02:
Firstly I don't work for Prusa, not my github lol :)

The vendor profiles are stored in the vendor ini files and cant be written to/saved.  Well they can be edited manually but the next config update from Prusa will overwrite any changes you make so doing it is pointless.  The 'defaults' can change over time as and when Prusa decide to.  As such you can NOT rely on them to be identical over a period of time.  

No point exporting them if the import wont replace the ones on the system.

That is why I always save copies of my own profiles.  That way they can be exported and imported and I am guaranteed to get the same settings.  

Don't know what BotQueue is but for working on multiple machines or setting up a new one the simplest way even though its a total hack is to just copy the entire profile directory from one machine to another.  That way you don't have to run through the wizard on each new install to get whatever system ones you want.