prusa3d/PrusaSlicer #4822: Eject button should always be visible when removable media is inserted

JS3910, 2020-10-03:
### Version
2.2.0+ release version.

### Operating system type + version
Mac OS 10.15.6

### Behavior
Currently, if you have removable media inserted the "Export to SD card/flash drive" button is visible, but the "Remove device"/eject button doesn't appear until after you've exported a file.

For scatter-brained people like myself it would be more useful to have this button always available as long as a storage device is inserted. It would save time and energy when you accidentally insert the SD card again, insert the wrong card or any number of similar situations where you now have to tab out to the OS to eject the storage media. 

It's also more consistent behavior for the application to display all available interactions with storage media when they are available, and not arbitrarily gate one of them behind another action (exporting to the card).

_Is this a new feature request?_
Low-hanging fruit improvement.

lukasmatena, 2020-10-03:
Currently the Eject button is not there at all, it only appears in a notification that opens after you exported to the card - I believe it is consistent enough and no other buttons show and hide. I personally think that PrusaSlicer is a slicer, not a drive manager. And I don't think we need to allocate extra space for an Eject button for cases that someone inserts a wrong SD card and is too lazy to eject it normally. My opinion at least.

JS3910, 2020-10-03:
Currently as in development build/next release?

Because right now in the live release version you have:

Before slicing
[image]

After slicing
[image]

After saving to card
[image]

It would not be inconsistent, confusing, or take more space to have all three buttons visible at all times but greyed out when they aren't functional (can't save before slicing, and can't save or eject with no card in) - this is expected and normal behavior from a software GUI. A button that, like it is right now, shrinks to reveal the "next step" isn't 🙂. 

This would also follow the convention already established in the rest of the program since all the main toolbar buttons grey out when they don't have a valid operation. There's nowhere else that I can find in the entire program where buttons are entirely hidden when they aren't relevant.

It would just be presenting functionality that already exists in the software, in the space already allocated to it, in a slightly wider usage context.

fe60, 2020-11-18:
The eject button has been moved to the newly introduced notifications in 2.3.0-alpha3. I have posted a screenshot in https://github.com/prusa3d/PrusaSlicer/issues/4949#issuecomment-727554482

fe60, 2021-11-09:
The `Eject` button appears in the notification after a successful export since a while. I got used to this behavior, even though I voted for the "show always, grey it out" option (https://github.com/prusa3d/PrusaSlicer/issues/4949#issuecomment-718198777).

In addition I switched to shortcuts. Hence I don't care much about the design/placement of this button anymore.

@JS3910 What about you? Do you still want to get this changed or can this ticket be closed?

lukasmatena, 2021-11-09:
The change has been in there for more than a year now, there were no other complains as far as I remember. I believe this can indeed be closed.

closed 2021-11-09 by lukasmatena