prusa3d/PrusaSlicer #4826: Update built-in presets automatically checkbox isn't respected

JS3910, 2020-10-03:
### Version
2.2.0+ release version.

### Operating system type + version
Mac OS 10.15.6

### Behavior
When I started PS today, it asked me about updating the system profiles and it's not the first time, I checked my preferences and I have it set to automatically update these. Unfortunately I have no better reproduction steps than this.

_Is this a new feature request?_
No

lukasmatena, 2020-10-03:
This works as intended. If you deselect it, PrusaSlicer will not even offer the update. The tooltip on the option in Preferences says:
[image]

If we just updated the presets without any confirmation, someone else would complain how it's completely unacceptable to just change the presets behind the user's back, because it is the user who should have final word on things like that. I would actually agree.
Not to mention that if there was any problem during the update, it would be difficult to troubleshoot if the user didn't know that the update had happened.

JS3910, 2020-10-03:
That's perfectly sensible, but in this case the issue is instead that the option is very confusingly named. "Update" and "automatically" are not ambiguous terms usually, and to me this wording means "I will always have the latest without interaction".

"Automatically check for updated built-in presets" or some wording like this would be clearer.

closed 2026-09-29 by prusaslicer-bot[bot]