prusa3d/PrusaSlicer #4827: Show incompatible print and filament presets option isn't respected for system profiles

JS3910, 2020-10-03:
### Version
2.2.0+ release version.

### Operating system type + version
Mac OS 10.15.6

### Behavior
Even when "show incompatible print and filament presets" is checked, if a non-system printer profile is selected then all the system print and filament profiles are hidden. Additionally, the option doesn't allow to see incompatible non-system presets when a system preset is selected. In fact, clicking around like mad, I can't find any combination of settings or presets where this option does anything at all.

And in fact, profiles inherited from the system profiles are not hidden by this mechanism - and they can't be hidden at all unless you "detach" them because the normal dependency options aren't available. This is extremely inconsistent and annoying to say the least.

Expected behavior is, you know, to actually see profiles marked as incompatible when we've specifically asked for it regardless if they are system profiles or not.

_Is this a new feature request?_
No

rtyr, 2020-10-03:
"Show incompatible print and filament presets" parameter will only activate the flag icon, which is then available in print, filament and printer settings so you can easily enable/disable incompatible profiles.

Does it help? Can we close this issue? 

Step 1
[image]

Step 2 
[image]

JS3910, 2020-10-03:
That's tremendously unclear - "Show incompatible presets" is not an ambiguous statement. Currently the functionality would be better described as "Allow display of incompatible presets" or something similar. But I think enabling the enabling of incompatible presets is a bit redundant (and the meaning of the flag icon isn't exactly clear - I had no idea it was even there or what it does😆). 

I'd suggest to either:
- Make the option in the preferences do what it actually says.
- Make the flag icon clearer. There's no well defined symbol for incompatibility, so I'd say maybe an eyeball icon, which at least is well established for show/hide functionality.
- Rephrase the preferences option to describe the current functionality.
- Remove the preferences option entirely and always have the show/hide icon displayed.
- As above but only display the show/hide icon in "Expert" mode so that it's still hidden in the more simplified UI variants.

bubnikv, 2020-10-16:
The icon is there primarily as a debugging tool for us to troubleshoot profile dependency. It is now being used for unintended purposes.

ThePreviousOne, 2023-08-17:
> The icon is there primarily as a debugging tool for us to troubleshoot profile dependency. It is now being used for unintended purposes.

Maybe so, but that doesn't mean it's not a legitimate use case. For example I have a 0.15mm nozzle but can only seem to get decent prints with a 0.2mm first layer height so my print presets aren't compatible with my Printer.

closed 2026-09-29 by prusaslicer-bot[bot]