prusa3d/PrusaSlicer #4829: Error when trying to build the dependencies package

danoveb, 2020-10-03:
Have spent hours now trying to setup a development environment to compile prusaslicer.  I have followed the "Noob" guide, and when executing "msbuild /m ALL_BUILD.vcxproj" it ends up with the following error:

Build FAILED.

       "c:\Prusaslicer\PrusaSlicer-master\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "C:\Prusaslicer\PrusaSlicer-master\deps\build\deps.vcxproj" (default target) (3) ->
       "C:\Prusaslicer\PrusaSlicer-master\deps\build\dep_ZLIB.vcxproj" (default target) (8) ->
       (CustomBuild target) ->
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.ta
       rgets(234,5): error MSB6006: "cmd.exe" exited with code 9009. [C:\Prusaslicer\PrusaSlicer-master\deps\build\dep_
       ZLIB.vcxproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:19.63

Anyone that can help me out?

The-EG, 2020-10-13:
I've just built it all with a standard MS Visual Studio 2019 setup, but I used the Ninja generator instead (`cmake .. -G Ninja ...` and then `ninja` instead of `msbuild ...`).

I was getting similar, odd errors from both the 'Visual Studio 16 2019' and 'NMake Makefiles' generators, and trying to build from within Visual Studio gave an unhelpful 'failed due to logger issues' error.

fly2mars, 2021-03-16:
I have met the same problem. When I build dep_ZLIB.vcxproj separately, it prompts the command git is not found.

So just add git to PATH, then it worked.

paynterf, 2021-09-11:
Having the same problem with VS2019 Community Edition, following the # Step by Step Visual Studio 2019 Instructions in src\doc\How to build - Windows.md document.

Here's a screenshot showing the error:

[image]

lukasmatena, 2023-02-09:
Closing, unable to reproduce.

closed 2023-02-09 by lukasmatena