prusa3d/PrusaSlicer #4830: MK3s Firmware 3.9.1 Print Quality Issues.

StevenInc, 2020-10-03:
### Version
PrusaSlicer
Version:   2.2.0+
Build:     PrusaSlicer-2.2.0+-202003211132

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Catalina Version 10.15.7 (Build 19H2)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   2.1 INTEL-14.7.8
Vendor:       Intel Inc.
Renderer:     Intel(R) Iris(TM) Plus Graphics OpenGL Engine
GLSL version: 1.20

### Operating system type + version
Macintosh : 10.15.7 (19H2)

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3s
3.9.1 -3518

### Behavior
Print quality is very poor with exaggerated radiuses and large gaps on the fill.
See Picture: https://photos.app.goo.gl/6LivaY1QdtNBkLC99

Download the latest version of PrusaSlicer.
Download the latest version of MK3s Firmware.
Update Firmware on MK3s
Download the MK3s parts stl. files.
Import y-rod-holder.stl
Print Quality: 0.20 Quality
Filament: Prusament PETG
Printer: Original Prusa i3 MK3S
Infill 20% Grid.
Slice and export to SD Card

ochm, 2020-10-03:
Looks like too high K value?

rtyr, 2020-10-03:
This issue is better suited for firmware repository https://github.com/prusa3d/Prusa-Firmware. 

**Please provide your g-code, so we can check it.**
- Are you sure you have your first layer (liveZ) set correctly?
- You mentioned FW 3.9.1, does it mean that it is OK with older firmware?
- Is your printer stock? Is it standard 0.4mm nozzle?

Something similar can happen with extreme high K value, which is not the case with stock profiles. I tried it according to your steps (including incorrect model orientation).

[image]

StevenInc, 2020-10-05:
Thank you,
Next time I will post it to the firmware thread.
As an update.:
What happened actually is the silicone boot I had over the hot tore, and it was creating the pattern that looked like it was a k-value error.
When I removed the boot, the print was a lot closer to the expected output.

Thank you to those that responded to my post.
Steve Benjamin

closed 2020-10-05 by StevenInc