prusa3d/PrusaSlicer #4831: Closing PrusaSlicer closes print host upload queue even if currently uploading

Sembazuru, 2020-10-03:
### Version
PrusaSlicer release 2.2.0-win64

### Operating system type + version
Windows 10 Pro Version 2004 OS Build 19041.508

### 3D printer brand / version + firmware version (if known)
MK3S with FW v3.9.1

### Behavior
When sending gcode to printer host (in my case, OctoPrint v1.4.2 on RPi4 over WiFi) I don't know how long it will take and sometimes accidentally close PrusaSlicer before sending completes, especially for large gcodes. I'd like some way of reducing the possibility of this happening. Some ideas on how to help this (seed for developer discussion before actual implementation):

- Show a progress bar in the status line in the bottom of the PrusaSlicer window that is a combined progress for all currently uploading gcodes in the queue.
- A configuration setting in preferences to automatically open and pull focus the "Print Host Upload Queue" window.
- If there are any currently uploading files on the queue when closing PrusaSlicer give the option to either cancel the close operation, close the program once all the uploads in the queue finish, or close the program interrupting all (including stalled) uploads. If option to close once the queue is finished, open and pull focus the queue window to allow the user to cancel any stalled or unwanted uploads.

There may be other ways to handle this, but I wouldn't mind all three ideas were implemented.

#### Project File (.3MF) where problem occurs
Not project specific.

kocikdav, 2020-12-11:
Hello, thank you for reporting. There is a new dialog window introduced at b589b9ce9cb1c645ff1bb45e69b46ade769cba9c. This will not let you close PrusaSlicer without notice if any upload is active. Please test when next beta version (3) is released. we will be further improving information on uploads in the future.

closed 2020-12-11 by kocikdav

Sembazuru, 2020-12-13:
The functionality works very well. I like the auto-open of the queue when one cancels the closing action. But I think the wording might be made better (at least the English text).
[image]
The question "Stop them and continue anyway?" might be better worded as any of:
* "Stop uploads and close?"
* "Stop uploads and continue to close?"
* "Would you like to stop uploads and close the program?"

bubnikv, 2020-12-13:
Thanks, Chris. We are not native English speakers.
Which wording do you prefer?

Sembazuru, 2020-12-19:
Sorry for the delay in getting back to you. Yeah, I know you guys aren't native English speakers (though your English is much better than my 2nd language). I had asked my wife for a better wording (she corrects my grammar) and she came up with the 3rd option. If I were writing it I'd probably go with the 2nd one after one edit: "Stop uploads and continue closing?" I guess which one you use depends on how wordy or terse you want.