prusa3d/PrusaSlicer #4832: First Layer Temp set in FIilaments tab is not inserted into Gcode

sharktank72, 2020-10-03:
### Version
2.2.0

### Operating system type + version
Win 7 SP2

### 3D printer brand / version + firmware version (if known)
Gcode issue -  printer is irrelevant

### Behavior
Setting a First layer temp never gets inserted in the Gcode. Only if you have a temp in your start code does the print start with a temperature set. "Other layers temp" does kick on correctly.

#### Project File (.3MF) where problem occurs
3mf and Gcode in zipped file. First layer temp was set to 220 in the Filament tab. This does no show in the Gcode.
[FirstLayerTempIssue.zip](https://github.com/prusa3d/PrusaSlicer/files/5322354/FirstLayerTempIssue.zip)

rtyr, 2020-10-03:
Removing this line from the start g-code will help.

[image]

rtyr, 2020-10-05:
Can we close this issue?

closed 2020-10-08 by rtyr