prusa3d/PrusaSlicer #4835: Save preset dialog not wide enough / not resizable

rent-a-developer, 2020-10-04:
### Version
2.2.0+win64

### Operating system type + version
Windows 8.1 x64

### 3D printer brand / version + firmware version (if known)
Orignal Prusa MK3S

### Behavior
1. Goto "Printer Settings" tab.
2. Click on the "Save current Printer Settings" button.
The "Save preset" Dialog opens.
Unfortunately this dialog is not wide enough to display the full name of the preset (see attached screenshot).
To make things even worse the dialog is not resizable.

[image]

Please either make the Dialog have a higher default width or make the dialog resizable.

rtyr, 2020-10-05:
This is caused by your custom scaling setting in Windows 8.1.

See for example this article https://www.technorms.com/32123/automatic-dpi-scaling-in-windows-8-1 for more details.

How it looks with the default settings (Win10).
[image]

rent-a-developer, 2020-10-17:
Thanks @rtyr for the info!

However, I still think such dialogs should be resizable.
Personally I hate it with a passion when dialogs that potentially contain larger contents have a fixed size.

closed 2026-09-29 by prusaslicer-bot[bot]