prusa3d/PrusaSlicer #4837: Gap through a sliced lithophane, but only when placed at certain angles

yoyo42, 2020-10-05:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Pro 1903 x64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3 upgraded to MK3S

### Behavior
When slicing a lithophane I get a vertical gap right through it that is not present in the object view. The place where it happens is not especially thin. It does end up as a gap in the real object.

Rotating the lithophane some around the Z axis and reslicing makes the gap disappear without a trace, with no other changes

[image]

#### Project File (.3MF) where problem occurs
[bug-gap.zip](https://github.com/prusa3d/PrusaSlicer/files/5330483/bug-gap.zip)

rtyr, 2020-10-05:
It is most likely caused by broken model, repairing it should help. 

[image]

yoyo42, 2020-10-05:
That option is greyed out for me and no errors were reported by the slicer. I have used it on other models.

rtyr, 2020-10-05:
You can use for example MS 3DBuilder to repair the model.

[sienna.jpgW100H140T3V5B2A0C0NS_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/5330568/sienna.jpgW100H140T3V5B2A0C0NS_repaired.zip)

yoyo42, 2020-10-05:
I guess you're right, I can't make that one do it. Weird that it didn't show up until sliced.

Thanks

closed 2020-10-05 by yoyo42