prusa3d/PrusaSlicer #4840: Slicer do not start

sim-sut, 2020-10-06:
### Version
Version: 2.2.0

### 3D printer brand
ORIGINAL PRUSA I3 MK3S
### Behavior
I installed Prusa Slicer two weeks ago and everything worked well. Yesterday I wanted to work with it but i could not start the program. When I try to open it (even when I run as admin) nothing happens. I can see the Slicer for e few seconds in the task manager but then the status turn on suspended. I uninstalled the program several times and installed it new but I have still this problem. I have installed the program on two other computers but there everything work well. Do you have any idea what I can do to fix this?

lukasmatena, 2020-10-06:
I don't know. Could you try to run it from the command line with argument `--loglevel=5` to see if it says something before quitting?

sim-sut, 2020-10-06:
Thanks a lot for the quick reply, could you tell me how I should use the `--loglevel=5` command with the windows executable (prusa-slicer.exe)? Please mind the fact that I am operating under Windows 10 64bit.

rtyr, 2020-10-06:
I will try to describe it.

**Run Windows explorer and navigate to PrusaSlicer folder (this is default location)**
[image]

**Click here, then type "cmd" and hit Enter**
[image]

**Type prusa-slicer-console.exe --loglevel=5 and hit Enter**
[image]

This last step will start PrusaSlicer (or at least attempt to start it), while the black command line window will be still opened. Then select all the text in the command line window (CTRL+A), copy it to the clipboard (CTRL+C) and paste it here (or to some txt file) using CTRL+V.

I am not sure if this is the easiest way but I hope it helps :-).

sim-sut, 2020-10-06:
[image]

Thanks for the instruction. You see in the picture what happens when I use this command. The version I downloaded is the newest available version for Windows 10. Do you have any idea what the problem could be?

lukasmatena, 2020-10-06:
@sim-sut I have no clue what's wrong, the error message is not very informative. But maybe some of the Windows folks among us will read something in between the lines.

rtyr, 2020-10-06:
@sim-sut I am not sure, there are some tutorials on how to get rid of such message. Maybe you can try to temporarily disable your antivirus. Are you aware of any recent windows update?

sim-sut, 2020-10-06:
I made all necessary updates. The only available installation now is the feature update to windows 10, version 2004 (current version 1909).

bubnikv, 2020-10-07:
We have one of our co-workers at Prusa Research complaining about a similar
issue, we will investigate on his computer.

bubnikv, 2020-10-16:
Windows 64 bit build on a Windows 32bit or Windows on ARM?

sim-sut, 2020-10-16:
I am not sure what you mean. I am operating under Windows 10 64bit.

bubnikv, 2020-10-16:
@sim-sut 
> I am not sure what you mean. I am operating under Windows 10 64bit.

Isn't it Windows 10s?

This is a weird kind of message. It looks as if Window did not try to start PrusaSlicer at all, it just looked into the application manifest, signature or whatever it did.

sim-sut, 2020-10-16:
No it is Windows 10 Pro with a 64-bit operating system and a x64-based processor. Yes excactly it looks like windwos does not want to start PrusaSlicer. I can see the name of the program in the Taskmanager. But sometimes the name disappears after a few seconds and sometimes I can see the name as long as I do not close the program in the Taskmanager.

bubnikv, 2020-12-05:
Do you have this issue with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-beta1
or with the beta2 (to be released soon)?

bubnikv, 2021-04-14:
No response, closing.

closed 2021-04-14 by bubnikv