prusa3d/PrusaSlicer #4842: E0 ant E1 temperature

PaaXus, 2020-10-07:
### Version
_2.2.0+win64_

### Operating system type + version
WIN10

When i set 2 extruders (i have chimera ext.) the temperature for E1 did not set in g-code. I set E0 for PETG (240C) and E1 for PLA (200C). I must set temperature only in custom filament g-code.

[image]

[image]

Is it formal function? Why after slicing g-code did not respect E1 temperature settings?

[image]
[image]

Thanx.

schnello, 2020-10-09:
Sliced with 2.2.0 and temp is set correct.

rtyr, 2020-10-09:
Please provide your 3MF project file.

PaaXus, 2020-10-12:
http://leteckaposta.cz/225001242

rtyr, 2020-10-12:
The issue is caused by your start g-code.

How to fix it:
- remove M104 and M109 lines from the start g-code completely (PS will insert correct M104/M109 commands automatically)
**or**
- modify M104/M109 lines in the start g-code (the tools must be specified in this case)
Example:
M104 T0 S{first_layer_temperature[0]} ; set extruder temp T0
M104 T1 S{first_layer_temperature[1]} ; set extruder temp T1
M109 T0 S{first_layer_temperature[0]} ; wait for extruder temp T0
M109 T1 S{first_layer_temperature[1]} ; wait for extruder temp T1

PaaXus, 2020-10-12:
OK, i will try it and i respond. Thanx!

rtyr, 2020-10-14:
Can we close this?

PaaXus, 2020-10-14:
Not yet, i will try it tomorrow. Thanx.

PaaXus, 2020-10-14:
You can close this. Thany you and cheers.

closed 2020-10-14 by rtyr