prusa3d/PrusaSlicer #4844: Ejecting an SD card is slow and locks UI

BeansIsFat, 2020-10-07:
### Version
2.2.0+

### Operating system type + version
Mac OS 10.14.6

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
Ejecting an SD card takes a long time and locks the UI.

Steps to reproduce:
* Insert SD card
* Export G-code to SD card
* Click Remove Device button

Expected result:
* Quick eject
* Responsive UI

Actual result:
* Slow eject
* UI non-responsive
* Old-style watch cursor
* Spinning beachball cursor

#### Project File (.3MF) where problem occurs
N/A

It does not appear to be an issue with my computer as ejecting via Finder is almost instantaneous.

bubnikv, 2020-10-08:
Sorry, but it is slow even in OSX finder.

BeansIsFat, 2020-10-08:
> Sorry, but it is slow even in OSX finder. 

On my computer it's quite quick to eject with Finder.

Nevertheless, if I've read the code correctly it looks like PrusaSlicer is calling the following command on Mac OS.
```
diskutil eject /path/to/removable/media
```
Using that command I can provide the timing from the shell.
```
$ time diskutil eject /Volumes/NO\ NAME
Disk /Volumes/NO NAME ejected
diskutil eject /Volumes/NO\ NAME  0.01s user 0.02s system 5% cpu 0.619 total
```
Since PrusaSlicer is significantly slower than this to eject it appears that there is something in the code beyond this command that is causing the issue.

brettvitaz, 2020-10-12:
Thank you for providing the feature. 

My experience is similar to @BeansIsFat. Ejecting via Finder or other methods is near instantaneous for faster SD cards and USB sticks. Ejecting using PrusaSlicer on MacOS adds so much time that I always eject using Finder.

mathiasvr, 2020-10-17:
I have always used finder to eject since it takes forever in PrusaSlicer so I can confirm this is an issue.

Le0Michine, 2020-10-19:
Got the same issue with prusaslicer and its relative superslicer on macos. I have several sd cards and the cheapest ones take same time to eject with prusaslicer and with finder (long time), whereas the good ones (sandisk) eject instantaneously via finder or cura but take forever with prusaslicer.

Olafejs, 2020-11-01:
I have the same issue on Mac mini ;/

koisdigala, 2020-12-19:
Issue exists for me on Slicer 2.2.0, on 10.15.7.  Using various SD cards in the built-in reader on the MBP.
It's so slow that I never use the eject feature inside Slicer.   :(

bubnikv, 2021-01-05:
Is eject in Cura faster then PrusaSlicer for anybody?

We are out of our wits.

mathiasvr, 2021-01-05:
@bubnikv For me Cura has always ejected similar to finder, so yes it is much faster. Unless the SD drive is busy it usually ejects instantly.

BeansIsFat, 2021-01-05:
I can confirm mathiasvr's experience. Just tested and the Cura eject speed is equivalent to Finder and significantly faster than PrusaSlicer.

I think the fact that old-style watch cursor first appears when ejecting with PrusaSlicer is an important clue. Usually that indicates a Carbon API call or some other non-native API. Since diskutil is native and the equivalent command is being used by Cura, it has to be something prior to that command, and likely a non-native API.

A quick look at RemovableDriveManager.cpp it seems to point to either the oneTBB or Boost libraries. Unfortunately, I'm not familiar enough with either of them to be of more assistance.

bubnikv, 2021-01-06:
We are using the same diskutil invocation as Cura. As far as I remember,
the diskutil invocation hangs.

I have the new Macbook Air with ARM processor here, I will test.

bubnikv, 2021-01-06:
@BeansIsFat

A simple
system("/usr/sbin/diskutil eject path_to_volume") blocks on my computer for 9 seconds when started from PrusaSlicer running under XCode. It is just the system() function call that blocks, nothing before or after it.

If I run the same command from command line, it executes in less than a second.

Now please advise.

From a sheer despair, we may try the convoluted method described in 
https://developer.apple.com/library/archive/documentation/DriversKernelHardware/Conceptual/DiskArbitrationProgGuide/ManipulatingDisks/ManipulatingDisks.html
at the end, section "Ejecting a Disk"

BTW I am a 3rd programmer at Prusa Research looking into this PrusaSlicer OSX eject issue. It is magic.

koisdigala, 2021-01-06:
Perhaps try debugging by invoking the eject in Slicer without doing the preceding File I/O to browse that fs and write the gcode file?  So don't query against the USB/SD card or write to it, just eject.

Maybe related to how you are detecting that removable storage was attached?

BeansIsFat, 2021-01-06:
@bubnikv 

Here's the most basic example I can come up with.
```
#include<stdlib.h>

int main()
{
    system("/usr/sbin/diskutil eject /Volumes/boot");
}
```
I compiled it with g++ then ran it from the command line with the following results.

```
➜  eject-test time ./a.out
Disk /Volumes/NO NAME ejected
./a.out  0.02s user 0.03s system 5% cpu 0.779 total
```
To me, this indicates that the system call is not, by itself, the cause of the delay. If it's blocking with just that system call in PrusaSlicer then it must be something in the application framework being used.

bubnikv, 2021-01-07:
I made a progress. After various experiments, I thought the only difference between PrusaSlicer and Cura may be, that Cura starts the tool from a worker thread, not from the main thread. I tried that, while blocking the main thread until the worker thread finished. That did not help. But if I just launched the worker thread while pumping the Cocoa messages (or whatever the wxWidgets do encapsulate Cocoa framework) in the main thread without waiting for the worker thread, the eject works very quickly.

So there is a light at the end of the tunnel. I have no idea what is going on. My hypothesis is that the diskutil tool is actually a Cocoa application and it somehow gets blocked by PrusaSlicer not interacting with the Cocoa framework while waiting for diskutil.

bubnikv, 2021-01-07:
I have little experience with Cocoa, but I have quite an overview of Win32. Win32 thread may or may not be a GUI thread. Maybe there is such a distinction in OSX/Cocoa as well?

bubnikv, 2021-01-07:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/18cf1fdb43cd6940733fdda8a4c1b07adff23b0e

closed 2021-01-07 by bubnikv

hubaskidoo, 2021-01-09:
I just tested on a Mac Mini m1 and see a 16 second lag before Prusaslicer responds to an eject command. Ejecting from the Finder is instantaneous.

koisdigala, 2021-01-09:
@hubaskidoo - did you test 2.3.0rc3?  I think it is fixed there.

hubaskidoo, 2021-01-09:
Yes. I am on rc3.

koisdigala, 2021-01-12:
Works well for me on 2.3.0rc3, thanks!

hubaskidoo, 2021-01-12:
On which Mac? Intel or Apple m1?

mathiasvr, 2021-01-12:
For me eject it's now working on my intel based MacBook 🎉

hubaskidoo, 2021-01-12:
That makes sense. I’m on an m1. If the binary is intel only, the eject is
being interpreted by Rosetta.

JustinFarris, 2022-01-18:
Running 2.4.0 on an Intel Mac Mini and I still have this issue. The eject hangs for 10+ seconds then fails. Any diagnostics info I should add?

mathiasvr, 2022-01-18:
@JustinFarris have you tried another SD card? I don't have this issue anymore.

JustinFarris, 2022-01-18:
Actually, no, in fact I've only tried with the USB drive that came with my Mini+. I'll try again with some others, but if my issue being USB delegitimizes my necroposting let me know.

BeansIsFat, 2022-01-18:
I just tried with a USB drive on my Intel MacBook Pro and it ejected quickly on 2.4.0 so I don't think it's a USB vs SD card issue.

What happens when you run `diskutil eject /path/to/removable/media` from the command line?