prusa3d/PrusaSlicer #4846: Tool change deactivates linear/pressure advance for the rest of the print (M900 K0)

Quas7, 2020-10-08:
### Version
2.2.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
**Marlin** 2.0.x-bugfix

### Behavior
For dual-extruders (with single nozzle) slic3r PE is sending a M900 K0 and deactivtes linear advance in Marin.
But it never recovers the K value later.

...
;--------------------
; CP TOOLCHANGE START
; toolchange #107
; material : PET -> PET
;--------------------
M220 B
M220 S100
; CP TOOLCHANGE UNLOAD
G1  X78.718 Y21.259  
**M900 K0**
G1 F2000
G1  X78.285 Y21.009  F2400
G4 S0
; Filament-specific end gcode 
;END gcode for filament
...

Is this a new feature request?
Could be a selectable option as a feature, if the K-value can be set in Slic3r PE.

#### Workaround ####
As tool-change script add: 
M900 K0.xyz   ;set linear advance

#### Project File (.3MF) where problem occurs
not relevant

rtyr, 2020-10-08:
It is expected, that "M900 Kxxx" command is inserted to the custom g-code section of the filament profile (different filament = different K). Do you have M900 command in your filament profile?

Quas7, 2020-10-08:
No, I did not add it to filaments as I use same filaments for multiple printers with different extruders. 
But it is clear for me, that this has to be done semi-manually.

I would still opt to have in "Advanced" a simple text field for K maybe in filament (override by printer?) to make it a bit maybe more straight forward for less experience (or lazy) users. ;)

closed 2020-10-08 by Quas7