prusa3d/PrusaSlicer #4848: Ghost stl on bed, cannot delete....

alnperez, 2020-10-08:
### Version
_Version of PrusaSlicer used goes here_  2.2.0 win64

_Use `About->About PrusaSlicer` for release versions_
Version: 2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,121MB
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version  
_What OS are you using, and state any version #s_  w10 2004 
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
an old stf file appairs An old ghost stl file appears systematically, it cannot be deleted and does not otherwise influence the gcode.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
green stl is à ghost...
_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2020-10-08:
That’s the purge block, it’s for purging on multicolour prints. I’m not at my computer at the moment but if you turn it off it will disappear

rtyr, 2020-10-08:
It is just a visualization of the wipe tower. PrusaSlicer doesn't have exact details about the wipe tower before slicing. It will change its size after slicing or completely disappear if there are no tool changes. Closing.

closed 2020-10-08 by rtyr