prusa3d/PrusaSlicer #4850: Brim sticking too well - request distance setting on expert mode

HeartOfGermany, 2020-10-08:
### Version
_2.2.0_

### Operating system type + version
_Windows 10 (unrelated)_

### 3D printer brand / version + firmware version (if known)
_Stock Ender 3 Firmware with custom high speed Prusa profile (also unrelated)_

### Behavior
* _The Brim ist often connected so strong to the part, that it is only cleanly removable with a lot of toolwork. The bed ist leveled almost perfectly (for my usecase it actually is absolutely perfect at the moment)._
### How to recreate
* _Print with brim enabled - happens on absolutely every print with uninterupted brim (not the spaced pattern coming from support)_
### Expected Results
* _Have the ability to change the brim offset from 0 to a value like 0.1 or 0.2mm_
* _This makes it possible to snap off the brim easily_
### Actual Results
* _Brim is fused so well, that a break-off without cleanup with tools is impossible_

Note, that the adhesion might be lower with less squish by increasing the nozzle-bed distance. However I intentionally use a tiny bit of squish, because the spaced pattern for support sticks like a steak on teflon. Means not at all. (Why is this pattern so horrible and creates small dots for example, which often get pulled off by the nozzle? Why is there no cross pattern, so the lines are at least somewhat connected and yield at least some interconnection stability?) Also, this tiny bit of squish also creates some other preferable effects like:
- Better bed adhesion in general
- smoother bottom surface
- much stronger bottom layer

Elefant foot compensation is no solution for obvious reasons. Neither is it intended for that purpouse, nor does it solve it, since layer 2 still connects with the brim, especially if it is a thin layer.

### Is this a new feature request?
* _I was unable to find something related, so I guess it actually is._

#### Project File (.3MF) where problem occurs
* _not needed, since it is a standard problem unrelated to specific files_

One might argue, that I am not supposed to squish the bottom layer. However this is giving me very great results. The lines are still printed extremely clean and a solid bottom layer actually looks fantastic to say the least. It might even be the official way, but there also might be many differing opinions on the topic.
One also might argue, that this is opposed the simplicity. It actually is not for 2 reasons. First, there are not a lot of options in the category "Skirt and Brim". Second, make it only visible on expert mode and fixed.

Please feel free, to discuss on this, world. Have a nice time.

BeatSlayer, 2020-10-08:
Brim is supposed to be mildly fused, you should try a skirt for more separation from object.

neophyl, 2020-10-08:
Normally with the Prusa Slicer method is to increase your elephants foot value to shrink your first layer perimeters to counteract the squish. This moves it in from the brim slightly. Of course if you move it in too much then the brim doesn’t connect at all and so serves no purpose. It’s something you have to tune for your first layer. 
I prefer the way it’s done in Super Slicer which is a brim offset like you suggest.

fe60, 2020-12-23:
Sounds like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3779

djen-k, 2021-01-28:
I would also really like to see a feature like this! I believe it should be really simple to implement and could be hidden as an advanced/expert feature so as not to confuse inexperienced users. It's such a simple thing that could go a long way to make the brim separate more easily and cleanly.

lukasmatena, 2021-08-12:
Brim offset will be configurable in 2.4.0-alpha1, which will be released soon.
Closing.

closed 2021-08-12 by lukasmatena