prusa3d/PrusaSlicer #4851: Adaptive perimeter support (very easy to implement) request: 1-2-1-2... perimeters

HeartOfGermany, 2020-10-08:
### Version
_2.2.0_

### Operating system type + version
_Windows 10 (unrelated)_

### 3D printer brand / version + firmware version (if known)
_Stock Ender 3 Firmware with custom high speed Prusa profile (also unrelated)_

### Behavior
* _The ammount of perimeters is fixed throughout the print/part/modifier element_
### How to recreate
* _Print anything_
### Expected Results
* _Switching from x perimeters to x+1 perimeters connects perimeters and infill much stronger. This is more than just an addition to "infill perimeter overlap"_
* _This makes super duper extremely strong parts possible within the range of material properties_
* _Layers would tear apart long before the infill disconnects from the perimeters, but this gives opportunity for very nice surface, since we do not need to crank up the infill overlap, which will result in much lower swelling of the perimeters_
### Actual Results
* _Perimeters are fixed, only way to really fuse infill and perimeters is by overlapping_
* _Reduced surface quality with high but required overlapping, because of filament spill or how one would call this_

Way(s) to fix it:
* _Perimeter count: 2,5 will print 2 and 3 perimeters alternately and embedd infill within the perimeters_
* _Use min and max perimeter count and also define intervall for more advanced slicing, though not neccesary in my opinion. Maybe someone would love to alternate between 2 and 4 perimeters for example though?_

### Is this a new feature request?
* _I am sure. I only have seen something about that topic once from another slicer actually._

#### Project File (.3MF) where problem occurs
* _not needed, since it is a standard "problem" unrelated to specific files_

Would totally love to see this feature at some time soon. This would really enhance the possibilities for strong functional parts without cheaping out on print quality.

Please feel free, to discuss on this, world. Have a nice time.

neophyl, 2020-10-08:
Super Slicer (a fork of PS) has an alternating extra perimeter option that locks in the infill better already.  So it should be easy to implement. It’s my default setting when using Suse.

bubnikv, 2020-12-07:
I believe this feature has been introduced by Johnathan of Kisslicer, and later implemented by Cura. I personally don't think that it will bring anything, and it could introduce banding artifacts on the outer wall. However it is straightforward to implement.

HeartOfGermany, 2020-12-07:
bubnikv, where is this feature hidden?

Anyways, this alternating perimeter really locks the features together much better. It really would be amazing, to have this for lightweight, high strength parts. Tuning slicer settings really can drastically increase the strength and print time. Prusaslicer is not able to eastimate my prints anymore. When it says 31 days, it finishes in like 2 days and 10 hours. :D And the part actually looks very reasonable for the absolutely insame time saving. And yet the part is strong as heck. But forcing overlap on infill and perimeters will create a much worse look than alternating the perimeter thicknes. For better optical result:
a) Print 3 perimeters (Px) like that: P2, P3, P1 (where the largest number corresponds to the outermost perimeter). This will alternate to P2, P3 and than again P2, P3, P1. This would make the artifact practically invisible, if it even would exist. It is a compromise of "Printing outside perimeters first" and alternating perimeters, without even printing the outside perimeters first. And the travel with aligned seams would be negligable.

rtyr, 2021-02-24:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/1011.

bubnikv, 2021-06-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1011, closing this one

closed 2021-06-03 by bubnikv