prusa3d/PrusaSlicer #4852: [Feature Request] "Solid Infill Last" option

Bpendragon, 2020-10-08:
An option to force all solid (including bridging) infill to be run last would be super useful, especially on the first layer of top solid infill as it would give the free-hanging filament something to maybe hold on to, instead of doing part of a solid block, suddenly stopping to go do sparse infill and then coming back to finish the solid infill, with the previous edge potentially having drooped due to lack of support. 

_Is this a new feature request?_
Yes

Luberry, 2020-10-14:
Either that or possibly add a perimeter would be useful like have a line or two for the bridge over the infill to hold onto.

bubnikv, 2020-12-07:
Ideally the infills would be ordered outside-in, starting with anything already printed to attach to.

closed 2026-09-29 by prusaslicer-bot[bot]