prusa3d/PrusaSlicer #4855: Feature request: Smart Bridging and Overhang

HeartOfGermany, 2020-10-09:
### Version
_any version_

### Operating system type + version
_Any operating system_

### 3D printer brand / version + firmware version (if known)
_Any FDM type printer_

### Behavior
* _Bridging is not handled smart_
* _Overhangs are handled worse_

### Expected Results
* _Having an option to really change the print speed of bridges and overhang_
* _Actually having the ability to define an overlap of the lines._
* _Printing unsupported overhangs with ease: I have written some code by hand to test, if an overhang without any support parallel to the print surface is possible. It actually is and it is not bad at all! At least it is way better than any overhang I have archived with any slicer so far. The top secret method I was using was as described below!_

### Actual Results
* _The lower solid infill is just sagging down and not connected to the perimeters_
* _Or it is fused to the support structure, if a solid layer below was used (I only use rectilinear grid with less than 0.2mm width as the setting for ease of removal for functional parts, works surprisingly well for what it is!)_
* _Support with shell breaks the actual part on removal, because it is fused far to much (for small [x,y] and at the same time tall [z] features like a standing bracket)_

### Proposed solution for overhangs and mid air infill
* 1. _Start printing from the inside of the part (which is somewhat solid), speed needs to be extremely slow like 10mm/s at absolute max for this to work! This is not really a problem, since this only happens for this overhang, so it basically does not add time, even if printing on a high speed profile. Fan: 100% (!!!!!)_
* 2. _Each line is printed with a constant offset from the previous line. For each new line this increases the distance 50% of the line width. (33% for even better finish required!)_
* 3. _With an offset of -3 lines, print a line with the same path as 3 lines earlier with z 0.1mm above (so actually 1.5 or 1 lines because we overlap, should be a varriable setting) . This layer on top will prevent sagging of the extremely thin floor we are printing. Floor stands for the overhang._
* 4. _If the full area is printed, finish with the upper support layer to the edge and than continue printing as usual and avoid the upper support layer till the next layer, to prevent collision, since it already is printed (if using different layer hights, this needs to be respected especially)_

The only problem I had was a tiny bit of sag on the end of each move. Everything else was as smooth as it could be possible. This likely can also be prevented by using a pull move continuing the path for a few mm with higher speed. Since it is hard to write such gcode by hand, I did not rework this. I was already very happy with the result.

 
### Example picture 1:
I used a thin box modifier to increase the perimeter count a lot. If we where able to print with overlap (even just 20%) and have it print these details very slow from the outside to the inside with fan at 100%, this would propably work perfectly fine without support. I mean, why should it not work? Hoewever I cannot modify the thin sheet box modifier to print in the order I want it to. Also the overlap is not possible at all.
[image]

Example picture 2 and 3:

This is a testcube I have printed with handwritten gcode. It is just a hollow cube with 2 perimeters and after finishing the last wall I started my code. The result is quite amating, what do you think? In the background a plaster copy of my finger for size comparision.

[image]
[image]

As you can see, the lines did sag a tiny bit at the end. This however definitely is fixable to a degree, since I did the wrong pull direction and did not calibrate anything. This was attempt number 7, actually. So I am quite proud, how great this worked.

### Related example gcode

; Code start |  
-- | --
; now printing overhang number: 25 |  
G1 X129.9 Y100.229 Z10 F2000 | ; Startpoint and z down
G1 E3.55 F4000 | ; Retract inverted
G1 X129.9 Y119.775 E0.2 F400 | ; Print bridge up
G1 E-3.5 F4000 | ; Retract
G1 X130.9 Y119.775 Z10.1 F2000 | ; Pull and Z up
G1 X129 Y115.775 F2000 | ; Backtrack
G1 X130.1 Y119.775 Z10 F2000 | ; Startpoint and z down
G1 E3.55 F4000 | ; Retract inverted
G1 X130.1 Y100.229 E0.2 F400 | ; Print bridge down
G1 E-3.5 F4000 | ; Retract
G1 X131.1 Y100.229 Z10.1 F2000 | ; Pull and Z up
G1 X129.2 Y104.229 F2000 | ; Backtrack
; change to top support mode |  
  |  
; now printing upper support layer number: 22 | overhang number: 25
G1 X128.7 Y100.229 Z10 F2000 | ; Startpoint and z down
G1 E3.55 F4000 | ; Retract inverted
G1 X128.7 Y119.775 E0.4 F400 | ; Print support layer up
G1 E-3.5 F4000 | ; Retract
G1 X129.7 Y119.775 Z10.1 F2000 | ; Pull and Z up
G1 X127.8 Y114.775 F2000 | ; Backtrack
G1 X128.9 Y119.775 Z10 F2000 | ; Startpoint and z down
G1 E3.55 F4000 | ; Retract inverted
G1 X128.9 Y100.229 E0.4 F400 | ; Print support layer down
G1 E-3.5 F4000 | ; Retract
G1 X129.9 Y100.229 Z10.1 F2000 | ; Pull and Z up
G1 X128 Y105.229 F2000 | ; Backtrack
; change to bottom bridge mode |  

### Explaination:

This was semi automatic with formulas in excel. I had the following variables to change, to generate different Gcode (no order, because it was a test only):

* Round (this is the only value that needs to be changed again and again starting from 0. The rest gets calculated and the generated gcode only gets copied and pasted again and again)
25.000
* Distance gain (this is back and forth, so the overlapp in this example is 0.2mm!)
.400
* Extrude mm (thin overhang layer, not optimized, but worked)
.200
* Print travel speed
400.000
* Retract speed (Is this even correct/safe? This was on stock Ender 3, so 25mm/s is the limit anyways)
4,000.000
* Move speed (move speed for non print moves like "pulling" and searching the next position)
2,000.000
* Round offset (after how many lines should top layer be started)
3.000

* Retract ammount
3.500
* Un-Retract ammount (compensating the lack of back pressure to make sure we have a  consistent flow)
3.550
* Z move (how thick the supporting layer above is printed)
.100
* Extrude 2 (thicker supporting layer on top)
.400
* Pull mm
1.000
* Back mm
.900

### Is this a new feature request?
_You can bet your legs on it. This technology in particular is propably world news!_

### important sidenote:

As the inventor of this technology I only allow this implementation in open scource projects like Prusaslicer and for myself if I want to do so. I am not kidding.

Please discuss, what you think on this missing feature. I mean, this would solve thousands of ugly prints and make them look almost perfect. There are no limits to implement this.

Thank you for your time,  dear reader.

IMightGitIt, 2020-10-14:
I have been thinking about the same issue, but have not been sure how to define it. Amazing investigative work!
For more standard anglede overhang, like an upside down pyramid, I think it would be really nice if the exterior perimeter was printed slower than normal. Properbly just a separate input setting, which could be a % of the normal print sprint.

HeartOfGermany, 2020-10-14:
And of course a slight overlap, to fuse the layers together, if the angle is to steep. Otherwise, it would definitely sag down or not fuse to the other layers well in the process. And if the overlap hinders, to reach the target angle, print another overlapping layer, to reach the goal.

I mean, sometimes I am even wondering, why there are no adaptive perimers. Seems quite obvious to me.

The problem likely is, that 3D printing is not a trend anymore. People can just buy a cheap Ender 3 and for 99% of the things on thingyverse any slicer is enought for most people. Companies are not seeking to invest money, if there is no big request for it. The ecosystem is pretty much built already, which hinders progress quite dramatically.

zsolt320i, 2020-10-24:
Very good ideas, i wanted to suggest similar things.
I also would three additional proposal:
1. To be able to setup/mark manually the bridging and overhangs areas. (i special cases it would be very useful).
2. To be able to define a distance **before** starting bridge printing where the **bridging speed** would have been already applied. 
E.g. If you would set 1mm, then with 1mm **before** bridging the printing speed would already **slowing down** to bridging print speed e.g.: 20mm/s.
3. To be able to define a distance before starting bridge printing where the cooling speed would have been already applied. 
E.g. If you would set 1mm, then with 1mm **before** bridging the cooling speed would already **going up** to bridging cooling speed e.g.: 80%.

Why it is important?
Because it is not possible to apply the bridge settings instantly at the nozzle, mainly at the printers with bowden tube.
Actually for me the begging of the bridging is the worst (Ender 3 with bowden tube), because at this moment the material is still flowing like crazy, and the cooling is still not efficient because the high cooling just started.

Pls consider for implementation.

Thank You!

closed 2026-09-29 by prusaslicer-bot[bot]