prusa3d/PrusaSlicer #4857: Print Host Upload Queue - column resizing issue

moong8te, 2020-10-10:
### Version
_Version of PrusaSlicer used goes here_
2.2.0+win64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 - 64 Bit - Build 2004
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
When resizing the columns in the Print Host Upload Queue for better visibility the headlines suddenly move all the way to the right and out of the window.
As there is no scroll bar to this window it leaves you with only the ID headline left.
At the same time the content of the columns seem to remain in place.
* _Steps needed to reproduce the problem_
Just drag the lines between the headlines to resize the columns and you should be able to observe the behavior.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[image]

moong8te, 2020-10-12:
I did a clean install on a different Windows 10 machine.
All details match the information in my original report above.
On the second machine I observe the same behavior and created a screen capture of it for better understanding of the issue.

https://youtu.be/WHR0fC1FJQY

As this seems quite simple to be reproduced does anyone else observe this behavior on his machine?

moong8te, 2020-10-17:
This issue should be easily reproducible even just by other users.
It would already greatly help if someone with the same version and Windows 10 64 Bit could or could not confirm this behavior.
As this could well be a matter of resolution, scaling etc. within Windows 10 I would very much appreciate any feedback.

bubnikv, 2020-12-07:
@kocikdav is looking into it.

closed 2026-09-29 by prusaslicer-bot[bot]