prusa3d/PrusaSlicer #4858: ColorFabb ,HT material preset outputs G-Code as PETG

rogerfroud, 2020-10-10:
### Version
_Version of PrusaSlicer used goes here_ 2.2.0+win64

_Use `About->About PrusaSlicer` for release versions_
Build 202003211145
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_

I'm selecting ColorFabb HT from the preset list but when I output the G-Code is says it's PETG

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ I'm expecting this to say ColorFabb HT. I've used this material before and I'm sure it said the right thing then. How do I know whether it's just the name that's wrong or whether it's going to ruin the print?
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-10-10:
Please post a 3MF.

rogerfroud, 2020-10-10:
[ColorFabb outputs gcode as PETG.zip](https://github.com/prusa3d/PrusaSlicer/files/5360241/ColorFabb.outputs.gcode.as.PETG.zip)

rtyr, 2020-10-11:
Why is it an issue in your oppinion? It is PETG filament type (copolyester).

cork3dprinting, 2020-10-11:
You can change the filament name that appears in the .gcode file name (while in advanced or expert mode) under Filament Settings: Advanced: Filament properties: Filament type. Hopefully this helps
[image]
[image]

rogerfroud, 2020-10-11:
The issue is that the name should reflect the filament I'm selecting. I
don't know what ColorFabb HT is made from, and the expectation is that this
is a profile that's been tuned specifically for it. If that's the case,
then the G-Code should say so and give confidence that it's not an error.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

rtyr, 2020-10-11:
We are using filament type in the name of the g-code file, not the filament name. Filament type is also used in the configuration wizard.