prusa3d/PrusaSlicer #4860: Settings Tab's Bold Font Regression

pmjdebruijn, 2020-10-11:
### Version
`2.3.0~alpha0+1079~g7826a202b` (current git master)
PrusaSlicer is built against the system copy of wxGTK 3.0.4

### Operating system type + version
Xubuntu 20.04 LTS

### Behavior
Vaguely recently, the text in the settings tabs have become bold, which reduces legibility and looks generally odd:

[image]

This may possibly only be an issue for wxGTK 3, otherwise I would have expected someone else to have already noticed this?

rtyr, 2020-10-11:
I wasn't able to reproduce it on Ubuntu 20.04 LTS / Win10 64bit (2.3.0-alpha+1078).

pmjdebruijn, 2020-10-12:
Any educated guess of what might be causing it?

FYI: I am building PrusaSlicer as a debian package locally, so with FHS enabled.

pmjdebruijn, 2020-10-12:
From what I can tell, it seems the regression must have been introduced between `2.3.0~alpha0+953~g8bf0f75e8` and `2.3.0~alpha0+1002~g0f44caa99`.

pmjdebruijn, 2020-10-13:
I tried bisecting, however that didn't go too well, given that the build is broken a lot of the time in between.

So after going through many of the commits I found this:

https://github.com/prusa3d/PrusaSlicer/commit/b15023dfa946682975a5bff8fada6fb187eab33f#diff-edf90ef33b3a80521aa3d4d4eba125ac99c885e9546968731cffe768e3d2c8dfR374 @bubnikv 

With the following resolving the issue, not sure if it positively or negatively affects OSX:

-			stb->SetFont(wxOSX ? wxGetApp().normal_font() : wxGetApp().bold_font());
+			stb->SetFont(wxOSX ? wxGetApp().bold_font() : wxGetApp().normal_font());

(A proper solution might end up being different of course, the above just proves that line of code is the root cause)

I also have a good guess as to why this is an issue for me, and seems just fine for many others:

PrusaSlicer uses 'Noto Sans', and bundles the Regular weight, but NOT the Bold weight.

Since I'm running Xubuntu which uses Noto Sans as it's default system font, I have a full family of Noto Sans available on my system, which wxGTK can use, even if there is no bundled Bold weight present with Prusa Slicer.

In cases where there is no system wide full family of Noto Sans, I would guess wxGTK falls back to it's bundled Regular weight, if no Bold weight is available, thus obscuring this issue from discovery.

So aside from adjusting the above line of code, I would suggesting that Prusa Slicer also bundles a copy of the Bold weight, so issues like this are less likely to remain unnoticed during development, and eventual release.

pmjdebruijn, 2020-10-26:
This issue seems to be only reproducable with wxWidgets 3.0.x, when rebuilding against wxWidgets 3.1.4 this issue has gone.

I would suggest ifdeffing the SetFont line, to only be applied with wxWidget > 3.0, having everything in not bold (including headers), is still an order of magnitude better than having everything in bold.

bubnikv, 2020-10-27:
@YuSanka would you please look into that?

bubnikv, 2020-12-07:
Is it still an issue, or could we close it?

pmjdebruijn, 2020-12-07:
This issue is no longer reproducable on Ubuntu 20.04 with Prusa Slicer `2.3.0~beta2+15~g82dfb990e` built against the standard wxWidgets 3.0.4 library shipped with Ubuntu 20.04. Presumably @YuSanka fixed this somwhere along the way.

closed 2020-12-07 by pmjdebruijn