prusa3d/PrusaSlicer #4861: ConfigWizard inconsistent printer names

pmjdebruijn, 2020-10-11:
### Version
`2.3.0~alpha0+1079~g7826a202b` (current git master)

### Operating system type + version
Xubuntu 20.04 LTS

### 3D printer brand / version + firmware version (if known)
Creality Ender 3

### Behavior
When running the ConfigWizard on first start (when ~/.PrusaSlicer-alpha does not yet exist), the Creality printers are listed under confusing/incomplete names.

When running the ConfigWizard on a second start (when ~/.PrusaSlicer-alpha already exists, and by explicitly requesting it from the menu), the Creality printers are listed by their proper names.

[image]

rtyr, 2020-10-11:
I wasn't able to reproduce it on Ubuntu 20.04 LTS / Win10 64bit (2.3.0-alpha+1078).

pmjdebruijn, 2020-10-12:
Any educated guess of what might be causing it?

FYI: I am building PrusaSlicer as a debian package locally, so with FHS enabled.

bubnikv, 2020-10-12:
case sensitivity of Linux file system?

pmjdebruijn, 2020-10-12:
Another interesting datapoint, would be that I only see this with Creality and TriLAB, not other the other third party brands, as Anycubic and Lulzbot seem fine. BIBO only has a single printer, so there's no obvious data there.

pmjdebruijn, 2020-10-13:
Also notice how 'Creality FFF' and 'Creality FFF Technology Printers' differs? Which seems to be respectively shortname vs title:

https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/ConfigWizard.cpp#L1927
```
pageFFF = new PagePrinters(q, vendor->name + " " +_(L("FFF Technology Printers")), vendor->name+" FFF", *vendor, 1, T_FFF);
```
Which calls:

https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/ConfigWizard.cpp#L471
```
PagePrinters::PagePrinters(ConfigWizard *parent,
    **wxString title,**
    **wxString shortname,**
    const VendorProfile &vendor,
    unsigned indent,
    Technology technology)
    : ConfigWizardPage(parent, std::move(title), std::move(shortname), indent)
    , technology(technology)
    , install(false)   // only used for 3rd party vendors
{
```

Which derives from:

https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/ConfigWizard.cpp#L395
```
ConfigWizardPage::ConfigWizardPage(ConfigWizard *parent, wxString title, wxString shortname, unsigned indent)
    : wxPanel(parent->p->hscroll)
    , parent(parent)
    , shortname(std::move(shortname))
    , indent(indent)
{
```

Which in turn derives from wxPanel.

What's unclear to me is how shortname and title get mixed up...

pmjdebruijn, 2020-10-26:
I just built PrusaSlicer with wxWidgets 3.1, and it's still reproducable.

pmjdebruijn, 2020-10-26:
After fiddling a bit with printer labels, I think this is due to wordwrap, where the affected labels, are too narrow causing premature wordwrap, and not high enough, causing words to disappear entirely.

So this might be vaguely related to #4911 / #4490

pmjdebruijn, 2020-12-02:
This issue still persist in current master.

pmjdebruijn, 2020-12-13:
@YuSanka this issue still exists in beta3

YuSanka, 2020-12-17:
Thanks for your report. This bug was fixed with 694082967603d12bb065d6b0ba47bec5719abe78 and will be a part of upcoming release.

closed 2020-12-17 by YuSanka

pmjdebruijn, 2020-12-17:
Confirmed fixed.