prusa3d/PrusaSlicer #4864: Using Tab in sidebar input fields causes sidebar to collapse

m-kozlowski, 2020-10-12:
### Version
version_2.3.0-alpha0-1108-gf0de7f13f

### Operating system type + version
Windows 10

### Behavior
* Using Tab to switch between input fields (i.e. scale factors) in sidebar collapses sidebar. No key modifiers are pressed, especially not shift;)

lukasmatena, 2020-10-12:
We know about it and plan to have a look at it tomorrow. It is somehow connected to recently merged https://github.com/prusa3d/PrusaSlicer/pull/4735. The issue is apparently Windows-specific, Linux and macOS seem to be ok. You can try to debug it, see GLCanvas.cpp:3219 and investigate why than code is executed.

m-kozlowski, 2020-10-12:
Well, it'll take me some time to dig into it, as I've never peek into slic3r/prusaslicer code before, but at the moment I'd say that the line you mentioned isn't executed. At least breakpoint does not trigger.

// BTW: shift+tab is bad choice IMO, as it's already used to toggle inputs in reverse direction.

lukasmatena, 2020-10-13:
> At least breakpoint does not trigger.

In that case I have an idea where the problem might be (inadvertent trigger of a menu item through an accelerator). Someone will fix it soon.