prusa3d/PrusaSlicer #4866: Incorrect layer number and layer height shown in Preview mode and incorrect M600 placement

Blom69, 2020-10-13:
### Version
Version 2.2.0+win64

Windows 10 build 1809

Prusa i3 clone with Marlin

### Behavior
In Preview mode the slider on the right display the incorrect layer number vs what is output to the gcode file in the custom gcode:
; #[layer_num]
; Z=[layer_z]

The layer number shown in the slider is 1 based where the [layer_num] variable is 0 (zero) based.
The layer height shown in the slider does not take any Z offsets into consideration and actually starts at -0.20 (for 0.2 layer setting)

For example, while the slider shows 1.30 (6) the custom gcode mentioned above outputs ; #5 and ; Z=1.9

I use these values to find the correct place to manually add the M600 filament change command.

The option to add a filament change via the slider is even worse affected.

In the above scenario it places the M600 at layer #3 so it is way off.

Other than these two issues I love the software.

bubnikv, 2020-12-07:
The layer ID in the G-code is zero based. I don't think this is an issue.

Would you please provide a 3MF to reproduce?

rtyr, 2022-03-10:
No response. Closing.

closed 2022-03-10 by rtyr