prusa3d/PrusaSlicer #4867: Interlock Rectilinear Infill

espr14, 2020-10-13:
### Version
v2.2.0

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
All

### Behavior
**Motivation:** All infills are printed in separate layers. If you break a part with 100% infill most likely it will break between these layers because they don't blend together. Still, you can interlock the layers to increase the strength.

**Solution:**
[image]

Intro: Print every odd line of the infill only but with 150% flow.

Detailed:
1. Print first layer of 100% infill print odd lines only with 100% flow
2. Stay in the same layer, print perpendicularly odd lines with 50% flow
3. Move to next layer
4. Print perimeters etc.
5. Print odd infill lines with 150% flow (direction as defined by usual rectilinear infill)
6. Print odd lines perpendicularly with 50% flow
7. Go to 3.

**Bonus:**
Split internal infill into to parts:
1. outer
2. inner
The outer can have higher (100%) infill while the inner can have lower (20%) infill rate. This is something like dynamic infill rate but this is easier to implement and will work for all infills immediatelly.

I've seen somewhere similar approach of filling deep gaps between walls. But that was in the same direction which reduces strength due to same direction. My proposal uses alternating directions.

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs

bubnikv, 2020-10-13:
This is an extremely wise idea. Please provide a pull request. Thank you.

espr14, 2020-10-14:
**Update:** Increase flow rate to 200% in step 5 and skip step 6. This should produce more consistent and even stronger result.

bubnikv, 2020-10-16:
Thank you for your suggestion.

closed 2020-10-16 by bubnikv

thierryzoller, 2020-10-19:
Why is this being closed ?

fiveangle, 2020-11-24:
This appears to be like a less sophisticated version of the more generic interlocking infill proposal I submitted here; https://github.com/prusa3d/PrusaSlicer/issues/4608

Am I missing something, or is this just a modification of the "combine-infill every 2 layers" option which would result in the model still breaking on layer boundaries, of which there would be half as many of them ?  What is the benefit of this suggestion vs #4608 ?

espr14, 2020-11-24:
It's not obvious from the image but lines in X direction will lock the odd-even layer gaps while lines in Y direction will lock even-odd layer gaps. "Infill every 2 layers" puts 2 layers above another 2 layers. It does not push it inside. Your idea is great but it creates gaps in Z planes.

fiveangle, 2020-11-24:
> Your idea is great but it creates gaps in Z planes.

It does not.

You are also not considering infill patterns beyond the most simplistic "line" version that I showed in the example pic. 100% Rectilinear infill with my proposal set to 2 layer combinations would result in the identical result as your except for the first 50% height partial first layer and last layer prior to top layers would be 100% and each intra-z overlap would be 200% vs 100% of layer height in your proposal.  Your idea is in fact the same as mine, as far as I can tell, except my idea overloads an existing function (the "infill every..." feature) for the purpose of 1) minimizing the code exposure that implementing the feature would require, 2) allow additional flexibility for the user to choose >50% intra-layer overlap (generally 100%-300%, but even 400% could be used if the user determines their printer is capable) and 3) not require validation that the FDM printer can successfully print 50% smaller than the smallest supported layer height.  Printing successively lower layer heights is typically impossible due to finite limit on printer mechanics, while printing successively higher layer heights is limited only by the material properties and the print speed commanded.

There are many ways to skin a cat, and in the end, determining if it's better to start with a half-layer and support only a single-layer worth of intra-z overlap or if supporting >1x layer height intra-z overlap as in my suggestion is better would need to be empirically determined.  Given the intention of the feature (to increase shear strength in the z axis for near-100% infill parts) I highly suspect using >1 layer height for intra-z overlap will be stronger.  This is why I did not submit the idea with a half-layer-height component.

Regardless, I took a look at the code and could not easily determine how to implement the feature myself so it will remain just an idea unless it piques the interest of an enterprising soul :)