prusa3d/PrusaSlicer #4868: Unable to Export GCode directly to SDCard

RSteel517, 2020-10-13:
### Version
_Version of PrusaSlicer used goes here_

 Version 2.2.0+win64

### Operating system type + version
_What OS are you using, and state any version #s_

Windows 10 64 bit 

### 3D printer brand / version + firmware version (if known)
n/a this is a slicer issue

### Behavior
* _Describe the problem_
After slicing a model, I am unable to directly export the G code to my removable memory card.  The card comes in as Drive E by default on my machine.  The export to Removable Media button shows up, as expected.

I have to save the G Code to my hard drive and then move it to the removable memory card.

* _Steps needed to reproduce the problem_
Open Prusa Slicer
Insert SDCard
Slice model
Export to SDCard

* _Expected Results_
GCode gets exported to SDCard

* _Actual Results_
Error message saying that it cannot export to SDCard

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

No

RSteel517, 2020-10-13:
[image]

bubnikv, 2020-10-13:
would you please upload the screenshot? It did not go through.

RSteel517, 2020-10-13:
[image]

RSteel517, 2020-10-13:
E:\ is the default drive that is given to the SD Card when i insert it into the computer.  I have checked to make sure that the 'Lock' tab on the SDCard is not in the 'Lock' position and have confirmed that I can read/write to the SD Card by navigating to it and deleting contents thru Windows explorer.

bubnikv, 2020-10-13:
Is there enough space left on that SD card?
Are you using any special characters in the file name?

RSteel517, 2020-10-13:
There is enough space on the card.  Using the same card, I am able to export the same GCode from Prusa Slicer onto my hard drive and then move that file onto the SD Card.  No name change.

The only special character might be the underscore ( _ )

bubnikv, 2020-10-14:
We are going to release 2.3.0-alpha1 any day now. Would you please test with the alpha?
If it is still not working, we will extend the alpha with some diagnostics that will help us to troubleshoot your problem.

Primermecos, 2020-10-17:
I had/have the same problem with my SD Card. My way "solving" this problem: I remove the SD Card and insert it right back in my card reader (sometimes, I have to remove it 2 - 3 times). 

Until just now, I was thinking that my memory card was slowly becoming unusable because it is old. That's why I bought a new one and will test it together with a new card reader.

My second idea: Could it be possible that Kaspersky is creating the problem? Most of the time, when I slide the card into the card reader, I get a message from Kaspersky that there is a problem on my memory card, which I ignore because I think there is no problem.
Perhaps the security software is blocking access to the SD card?

RSteel517, 2020-10-29:
I have not yet had the same issues with the new Alpha release.  It appears to be working correctly.  If I run into this again, I will reopen this issue.

Closing this issue.  Thank you.

closed 2020-10-29 by RSteel517