prusa3d/PrusaSlicer #4871: OpenGL 2.0 capable graphics driver error

BrerDane, 2020-10-14:
### Version
_Version 2.2.0+win64_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 Home 10.0.1904 Build 19041_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 16,946MB
OpenGL installation
GL version:   1.1.0
Vendor:       Microsoft Corporation
Renderer:     GDI Generic
GLSL version: N/A
Installed extensions:
GL_EXT_bgra
GL_EXT_paletted_texture
GL_WIN_swap_hint

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
_Prusa / i3 MK3S MMU2S

### Behavior
* _Describe the problem_

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
"PrusaSlicer requires OpenGL 2.0 capable graphics driver to run correctly
while OpenGL version 1.1.0 render GDI Generic vendor Microsoft Corporation was detected.
You may need to update you graphics card driver.
As a workaround you may run PrusaSlicer with a software rendered 3D graphic by running
prusa-slicer.exe with the --sw_renderer parameter.

Have NVDIA graphic card which has latest drivers and did not get this until this update. The --sw_renderer parameter does not work using target option in Windows shortcut. 

_Is this a new feature request?_
No, worked fine before.
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
N/A happens on launch of PrusaSlicer

bubnikv, 2020-10-14:
> Have NVDIA graphic card which has latest drivers and did not get this
until this update.

Until the update of what? Windows? NVIDIA drivers? PrusaSlicer?

> The --sw_renderer parameter does not work using target option in Windows
shortcut.

That should work, for example

https://www.digitalcitizen.life/shortcut-arguments-parameters-windows

TheLion, 2021-01-11:
I have a similar issue on my Microsoft Surface Book 2 when I attach two screens through a docking-station. When I start PrusaSlicer with the docking attached, I get the error:

```
PrusaSlicer requires OpenGL 2.0 capable graphics driver to run
correctly while OpenGL version 1.1.0 render GDI Generic vendor Microsoft
Corporation was detected.
You may need to update you graphics card driver.
As a workaround you may run PrusaSlicer with a software rendered 3D
graphic by running prusa-slicer.exe with the --sw_renderer parameter.
```
When I add "--sw_renderer" to the shortcut, PrusaSlicer doesn't start at all.

When I disconnect the docking station and start PrusaSlicer without the "--sw_renderer", it starts just fine. When I add "--sw_renderer", it doesn't start and reports no errors.

Also tried it on a PC instead of a laptop with two screens attached, where it works just fine without "--sw_renderer", but with "--sw_renderer" it doesn't start.

I'm currently using PrusaSlicer 2.3.0rc3, but I noticed this issue earlier, can't say exactly, but for sure with an earlier 2.x version.

I have the latest drivers for both my graphics card and my (Club-3D) docking station.

kaareseras, 2021-02-11:
> I have a similar issue on my Microsoft Surface Book 2 when I attach two screens through a docking-station. When I start PrusaSlicer with the docking attached, I get the error:
>
> ```
> PrusaSlicer [...]

I see the exact same behavior on my Surface book, also only with external screens

TheLion, 2021-02-12:
This also happens with v2.3.0 of PrusaSlicer.

Danafra, 2021-05-26:
I've just got this message as well for the first time this morning. This is with 2.3.1.

Bravehartk2, 2021-08-25:
Same Problem on  2.3.3 on Lenovo P1 Quadro T1000

Tetz95, 2021-09-08:
Can confirm this issue with an XPS laptop and a thunderbolt dock.

bubnikv, 2021-09-09:
It may be that your issue is not specific to PrusaSlicer.

Could you please try http://realtech-vr.com/home/glview
Cura or Ideamaker software,
https://openscad.org/
etc?

We have one machine here at PrusaResearch, where PrusaSlicer used to crash with a USB3 docking station connected with two large monitors attached. We later found out that no OpenGL application was running on that setup. Also interestingly enough it is a dual graphics card setup. If one of the monitors is disconnected, PrusaSlicer starts using the more powerful NVIDIA graphics card. One may then reattach the second monitor to the docking station and everything runs fine. The guy recently got an update and since then PrusaSlicer gets the Microsoft software OpenGL 1.1, where PrusaSlicer does  not run either.

The glview has a "rendering test" mode, where our machine described above crashed. OpenSCAD crashed after one started with "new project".

cagaroutte, 2021-11-24:
i'm using HP workstation with NVIDIA graphics card, set up with dual monitors.  I had an issue with Solidworks not opening when my laptop is closed.  Solidworks recently corrected itself with an update, but now, SuperSlicer won't open with the laptop closed.  I opened the laptop, Superslicer opens.  FYI>

TheLion, 2021-12-20:
With the same hardware setup as before (https://github.com/prusa3d/PrusaSlicer/issues/4871#issuecomment-758106742), it works with Windows 11!

bubnikv, 2021-12-20:
> i'm using HP workstation with NVIDIA graphics card, set up with dual monitors. I had an issue with Solidworks not opening when my laptop is closed. Solidworks recently corrected itself with an [...]

Please report to SuperSlicer.

iamjamieq, 2022-02-07:
Had this issue myself. Laptop with RTX 3060, PrusaSlicer worked perfectly at home. Get to work the next day and get this error. Difference: I use a USB 3.0 docking station at work, with my monitors at home plugged in directly by HDMI and Mini DP. So I unplugged my main monitor from the docking station and plugged it into the inboard HDMI port. PrusaSlicer opened. Hope this helps someone.

jdelucca, 2022-08-03:
I ran into this issue on a Microsoft Surface laptop and version 2.3.1. Like some of you, it was working perfectly fine one day and refused to open the next. 
My solution was to plug it in and reboot. Most likely it was switching from graphics card to integrated graphics, and will run the graphics card when plugged in.

closed 2026-09-29 by prusaslicer-bot[bot]